slic3r/Slic3r #2545: SIGSEGV while slicing https://youmagine.com/designs/modified-space-wrench

hyperair, 2015-01-15:
When slicing SpaceWrench2.stl from https://youmagine.com/designs/modified-space-wrench#!design-documents, Slic3r 1.2.5 running on Ubuntu 14.10 segfaults. Here's the backtrace:

    (gdb) bt
    #0  0x00007f30d64c0cd8 in Slic3r::ExtrusionEntityCollection::first_point (this=0x7f30a80028b0) at src/libslic3r/ExtrusionEntityCollection.cpp:50
    #1  0x00007f30d62b33e9 in XS_Slic3r__ExtrusionPath__Collection_first_point (my_perl=<optimised out>, cv=<optimised out>)
        at /usr/bin/perl -MExtUtils::XSpp::Cmd -e xspp -- -t "/build/slic3r-HJKVzu/slic3r-1.2.5+dfsg/xs/xsp/typemap.xspt"  "/build/slic3r-HJKVzu/slic3r-1.2.5+dfsg/xs/xsp/ExtrusionEntityCollection.xsp":71
    #2  0x00007f30d90d6f1b in Perl_pp_entersub () from /usr/lib/x86_64-linux-gnu/libperl.so.5.20
    #3  0x00007f30d90cf7b6 in Perl_runops_standard () from /usr/lib/x86_64-linux-gnu/libperl.so.5.20
    #4  0x00007f30d9058d55 in Perl_call_sv () from /usr/lib/x86_64-linux-gnu/libperl.so.5.20
    #5  0x00007f30d6f4ad0a in ?? () from /usr/lib/x86_64-linux-gnu/perl/5.20/auto/threads/threads.so
    #6  0x00007f30d852e0a5 in start_thread (arg=0x7f30b7fff700) at pthread_create.c:309
    #7  0x00007f30d8d4888d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
    (gdb) bt full
    #0  0x00007f30d64c0cd8 in Slic3r::ExtrusionEntityCollection::first_point (this=0x7f30a80028b0) at src/libslic3r/ExtrusionEntityCollection.cpp:50
    No locals.
    #1  0x00007f30d62b33e9 in XS_Slic3r__ExtrusionPath__Collection_first_point (my_perl=<optimised out>, cv=<optimised out>)
        at /usr/bin/perl -MExtUtils::XSpp::Cmd -e xspp -- -t "/build/slic3r-HJKVzu/slic3r-1.2.5+dfsg/xs/xsp/typemap.xspt"  "/build/slic3r-HJKVzu/slic3r-1.2.5+dfsg/xs/xsp/ExtrusionEntityCollection.xsp":71
            THIS = <optimised out>
            RETVAL = {
              val = 0x0
            }
            sp = <optimised out>
            ax = 156
    #2  0x00007f30d90d6f1b in Perl_pp_entersub () from /usr/lib/x86_64-linux-gnu/libperl.so.5.20
    No symbol table info available.
    #3  0x00007f30d90cf7b6 in Perl_runops_standard () from /usr/lib/x86_64-linux-gnu/libperl.so.5.20
    No symbol table info available.
    #4  0x00007f30d9058d55 in Perl_call_sv () from /usr/lib/x86_64-linux-gnu/libperl.so.5.20
    No symbol table info available.
    #5  0x00007f30d6f4ad0a in ?? () from /usr/lib/x86_64-linux-gnu/perl/5.20/auto/threads/threads.so
    No symbol table info available.
    #6  0x00007f30d852e0a5 in start_thread (arg=0x7f30b7fff700) at pthread_create.c:309
            __res = <optimised out>
            pd = 0x7f30b7fff700
            now = <optimised out>
            unwind_buf = {
              cancel_jmp_buf = {{
                  jmp_buf = {139847222163200, -8160057251535346315, 1, 85884256, 139847222163904, 139847222163200, 8115707349989238133, 8115512419116688757},
                  mask_was_saved = 0
                }},
              priv = {
                pad = {0x0, 0x0, 0x0, 0x0},
                data = {
                  prev = 0x0,
                  cleanup = 0x0,
                  canceltype = 0
                }
              }
            }
            not_first_call = <optimised out>
            pagesize_m1 = <optimised out>
            sp = <optimised out>
            freesize = <optimised out>
            __PRETTY_FUNCTION__ = "start_thread"
    #7  0x00007f30d8d4888d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
    No locals.

It looks like the cause of the segfault is that there is no first point, i.e. it's an empty ExtrusionEntityCollection:

    (gdb) p this->entities
    $3 = std::vector of length 0, capacity 0

hyperair, 2015-01-15:
Looks like 1.2.4 slices this correctly.

hyperair, 2015-01-15:
Oh, I forgot, here's my config: https://gist.github.com/b5ca50491555034f35a7

swsch, 2015-01-15:
I get the same backtrace with (at least) two of my stl files, while many others work fine. HEAD slices the files just fine.

alexrj, 2015-01-18:
Thank you, this was fixed.

closed 2015-01-18 by alexrj