slic3r/Slic3r #2547: combine multimaterial

alke, 2015-01-15:
Hello
I wonder why combine multimaterial is not in the file menu on version 1.2.x 
Ubuntu 14.04

Idont now if this is the right place to ask but try
Sinserly Kent

alexrj, 2015-01-15:
Because it was decided to remove it.

The same thing can be done with these steps:
* add the first part to the plater
* double click on it
* click "Load part…" to add the other part(s)
* close the dialog
* Export as AMF file (if you want)

closed 2015-01-15 by alexrj

peterjsp, 2015-06-25:
I can export to AMF using the above steps however when I try to reload the file in Slic3r I get the following error...
[image]