slic3r/Slic3r #2548: Feature Request: Triangular infill

braddo99, 2015-01-15:
I think triangular infill will be the strongest per %infill setting in addition to being not very demanding on effector movement.

Hexagonal, which seems to be the new default, would have similar strength characteristics but due to how it is created (long durations of tight zigzag motion) seems to unnecessarily cause high vibration. 

Triangular can be created with long straight paths "bouncing" off of the perimeter layers. Unlike rectilinear fill the effector rarely does 180 degree reversals so, in addition to being a stronger geometric pattern, it's also less demanding on the machine.

MatterControl currently has triangular infill and I have used and like it quite a bit.

alexrj, 2015-01-15:
Can you post a drawing?

braddo99, 2015-01-15:
Here's a picture, not from Matter control but same approach:

http://www.gyrobot.co.uk/blog/my-adventures-with-3d-printed-insoles-part-2-4

a4jp-com, 2015-01-16:
I would love a triangular infill as well ^^ It looks great and I could use it with lots of different models.

I don't see how a triangular pattern would reduce zig-zag motions though. It would just replace two vertical lines with one horizontal one, as far as I can see. 

P.S. Alessandro, I think Angus posted some code on your bug report to fix the lines crossing over.

braddo99, 2015-01-16:
You'd have to see it in action, imagine how you draw a five pointed star - your pencil traverses the entire area and when it reverses it's not 180 degrees. I'll slice something and gcode to check out.

braddo99, 2015-01-16:
Here's one from Craftware, the slicer mentioned in that link above. Hadn't used it before, but I think both it and MatterControl use Cura engine under the hood. I couldn't use MC without connecting it to a printer. That's neither here nor there I guess. Anyway, here's a gcode with the triangular fill. Note how the lines bounce around. Makes for a very strong and smooth fill.
https://drive.google.com/file/d/0Bxr_TT2OFDCET0w5TFVyTVBtMGM/view?usp=sharing

braddo99, 2015-03-22:
I would like to reemphasize that triangular seems to be a better default infill than hexagonal. It is far less strenuous for the bot to do fast infill (it is even less strenuous than linear infill since the reversals are not at 180 degrees) and it gives even higher strength than does hexagonal at an equivalent infill density. The current implementation of hexagonal uses very fast "zigzag" motions which can shake the bot apart at higher speeds. This is not necessary and not good IMO.

lordofhyphens, 2016-07-16:
@braddo99 well we'd have to re-implement it anyway because I don't believe the source for MC or Craftware exists. I think you can tweak Honeycomb.pm to turn it into Triangular fill if you want to mess around with it.

mrvn, 2016-07-18:
I can see this being done two ways:

1. Take rectilinear infill but rotate by 60° every layer instead of 90°.
2. Actually draw lines in all 3 orientations in every layer.
2a. Draw each direction separately, basically 1 but doing 3 infill layers at once.
2b. Reflect at the side giving a continious path for all 3 directions (or a few paths).

bubnikv, 2016-07-18:
The triangular infill would be quite simple to implement in Slic3r actually. Just copy/paste the grid infill and add the third direction. This will lead to a 3x overfill at the crossing points, but I bet this is what the other slicers do and it works well enough.

lordofhyphens, 2016-07-18:
Yeah, my experiments were trying to it to lay out similar to how the
Honeycomb works (which is a bit obtuse to work out how it works, at least
to me). I'll take a look at starting from Grid.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

bubnikv, 2016-07-18:
The way Simplify implemenents the rectilinear / grid / triangular infill is it gives a rectilinear infill, rotation angle and a number of repetitions. These values are freely configurable from the UI.

lordofhyphens, 2016-07-18:
Yeah, it's probably worth refactoring the fills; I'll give it a look
tonight.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2017-01-16:
Fixed via a881f755a1dc3a8b45a3a3a1f354297890fc3a08

closed 2017-01-16 by lordofhyphens