slic3r/Slic3r #2550: Skirt Adhesion.

sirkha, 2015-01-16:
Tall skirts don't like to stay on the bed for the whole print. This is caused by two things, I think. First, the skirt is very thin on the bottom unless you a large number of loops, and even then, with ABS-like materials, this results in shrinkage pulling it off the bed. To solve this, I would implement a first-layer skirt loops option that effectively gives the skirt a brim. The second issue with skirt-bed adhesion I am experiencing is with the multi-material first layer skirt. I understand the want/need to build pressure by generating a skirt loop with the second extruder on the first layer, but when using a different material, like Ninjaflex, in my case, the second layer skirt doesn't want to stick to it very well. I would solve this by always putting the non-skirt-extruder material in an additional loop outside where the skirt will print on the next layer.

This applies to all versions of slic3r with any model where you would want to print a tall skirt, with any other combination of settings.

lordofhyphens, 2016-07-11:
Could you go into more detail as to what the problem is with some pictures?  @alexrj  prefers to examine the problem before jumping straight to a solution.

pavltom, 2017-02-09:
Hi,

   I was having the very same problem: high skirt around an elongated object was detaching from the heatbed eventually ruining the whole print. My aim was to enlarge the brim width in slic3r, so it is wider than the skirt and thus the skirt can sit on the top of the brim. Unfortunately, I found out the effective skirt dimension is taken as a maximum from the skirt distance and the brim width, therefore the skirt never touches the brim.
   After some slic3r code browsing I have decided to modify it in the following way: if the brim is wider than the skirt distance, 2 skirt polygon collections are created. The usual "skirt" distant "skirt distance" from the object beginning from layer 1 and the new "brim skirt", distant "brim width" from the object, basically copying the brim with a tiny gap on layer 0 and the same placement as if the effective skirt size was taken from the brim width. If brim width is lesser than the skirt distance, no action is taken and only single ("skirt") polygon collection is made.
   This code change proved very useful and the skirt attached to the brim never break off again. If you like the concept, I am ready to create a pull request and possibly incorporate the changes in the master branch.
[image]

lordofhyphens, 2017-02-09:
You can always provide a pull request. Unfortunately I didn't understand what @sirkha was looking to solve and he never responded. 

I still would like a good description of the problem (with pictures of what is going on) before diving into solutions.

pavltom, 2017-02-10:
I have added the pull request, but I am afraid I am not able to provide any picture of my failed prints at the moment. Nevertheless, it's very easy to describe them: if the skirt is narrow and tall (like for the bolt on the 3D preview snapshot above), after some time the vibrations and hot end friction makes it detach from the heatbed. Once the skirt is not on its place, the printer lays filament into the air and ruins the print.

Also I should note I was printing with ABS, where the tension due to temperature contraction is much more significant and the skirt is more likely to detach. I guess PLA is not so prone to this issue.