slic3r/Slic3r #2552: Crash mostway through exporting Gcode (1.2.5.35, Win7)

natevw, 2015-01-16:
Slic3r 1.2 crashes when trying to export Gcode for http://www.thingiverse.com/download:745391 ('la-1001-r5.stl' of http://www.thingiverse.com/thing:444741/#files). I am rotating it -90º around X if that makes any difference.

```
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	slic3r.exe
  Application Version:	1.2.5.35
  Application Timestamp:	503fe867
  Fault Module Name:	XS.dll
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	54afd05d
  Exception Code:	c0000005
  Exception Offset:	000000000020f7e8
  OS Version:	6.1.7601.2.1.0.768.3
  Locale ID:	1033
  Additional Information 1:	53ca
  Additional Information 2:	53ca6a1eb975170a8d59149252e5999a
  Additional Information 3:	356d
  Additional Information 4:	356d51920ea0ce2a366b564297f6ad7f

Read our privacy statement online:
  http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0409

If the online privacy statement is not available, please read our privacy statement offline:
  C:\Windows\system32\en-US\erofflps.txt
```

Not sure if related to #2534 — different platform, and results in a 0 byte Gcode file instead.

natevw, 2015-01-16:
I was able to export this using 1.1.7 with no crash (remains to be seen whether it's worth printing at the unrotated orientation ;-)

alexrj, 2015-01-18:
1.2.6-dev doesn't crash using the instructions you provided, so I really think this was related to the issue that was recently fixed.

However note that the object is very small (about 1.3x1.3cm) and has very thin walls where there's no room for extrusions. You might want to scale the object before printing it.

alexrj, 2015-01-18:
[image]

closed 2015-01-18 by alexrj

natevw, 2015-01-22:
I'd agree that the crash was probably https://github.com/alexrj/Slic3r/issues/2517 and nice tip on the wall size. The model was made for a fancy inkjet+powder printer, so I wasn't holding my breath for a great part even if I could print it.

(Thanks for looking into these things so thoroughly, btw! Slic3r keeps getting better and better — once that constant random crashing fix gets out anyway ;-)