slic3r/Slic3r #2553: Gcode export stalls with "Object.pm line 293" message (Win7, 1.2.5)

natevw, 2015-01-16:
When trying to export Gcode for http://www.thingiverse.com/download:29561 ('chuck_wed_117.stl' file from http://www.thingiverse.com/thing:9243/#files) the process stalls. I noticed in the status bar it says "Usage: Slic3r::Layer::id(THIS) at C:/…/Slic3r/Print/Object.pm line 293." so perhaps a script is erroring out without getting detected?

Note that this file seems to have been exported from SketchUp and is "Auto-repaired (1296 errors)".

alexrj, 2015-01-18:
I fixed the issue causing that error, but that file is still broken and Slic3r is not able to repair it. Extending the cases where Slic3r tries to repair broken STL files is not in my current plans, so I'm not doing anything for this. The file must be repaired before being fed to Slic3r.

closed 2015-01-18 by alexrj

natevw, 2015-01-22:
Fair enough, makes sense to focus on the toolpath generation.

Though I will say: there's *soooo* many broken geometries on Thingiverse there must be some way their authors printed them without noticing! Do you know how $WHATEVER_THEY_USE deals with this, or have a favorite tool for repair? (So I'm glad Slic3r at least has some repair capability built-in, thanks!)