slic3r/Slic3r #2554: Crash after delete all object off plater [1.2.5]

natevw, 2015-01-17:
This has happened several times now, so I don't think it's coincidence:

1. Plate some objects and export GCode
2. Press "Delete All" to clear the plate and start some new stuff
3. That works, but soon afterwards: program abends!

alexrj, 2015-01-18:
I'm not able to reproduce this. Can you provide more details please?

* What is your operating system?
* Does this happen with any STL file and any combination of them?
* Does this happen with default config as well?
* Do you press the Delete All button when the background processing is still going on? If so, what's written in the toolbar right before you press the button?
* Does the system report any kind of crash log/stack trace?

natevw, 2015-01-22:
Hmm, I'm having trouble reproducing this too, and about the only question I can answer from memory is that it was Windows 7. I was primarily doing some rotation and then G-code export.

I'm willing to assume this was just another variant of https://github.com/alexrj/Slic3r/issues/2517. If I figure out a reliable way to get it to crash in a different way, I can file a new report.

closed 2015-01-22 by natevw