slic3r/Slic3r #2555: Perl script not working in Slic3r 1.1.7 with Repetier Host on Mac

Ohmarinus, 2015-01-17:
Hi guys, been trying to get this to work for ages, I found the script on the Reprap forums, but nobody can help me there. I'm clueless. In the IRC channel also nobody answered I waited a few hours in there.

Mac Mavericks with latest Perl installed and working.
Slic3r 1.1.7
Repetier Host 0.56

This perl script should work:

#!/usr/bin/perl -i.before_postproc
#
# Author  : Guy Poizat
# Version : 1.2
# Copyright : none.
#
# Change from previous version : added handling of slic3r's retraction speed setting,
#  so that this script is not disrupted by a setting change.

use strict;
use warnings;

# what will replace a retraction line with : gcode to cut the laser power off.
my $powerOffLine = "M104 S0\n";

# what will replace an "unretraction" line with : gcode to switch the laser power on.
my $powerOnLine = "M104 S100\n";

# Extract the speed setting from slic3r's environment variables (or use the default 30mm/s).
#  Unit is millimeters per minute in gcode, millimeters per second in slic3r configuration.
my $defaultRetractSpeed = 1800;
my $envRetractSpeed = $ENV{"SLIC3R_RETRACT_SPEED"};
my $retractSpeed = defined( $envRetractSpeed ) ? $envRetractSpeed * 60 : $defaultRetractSpeed;

# pattern of a retraction line - would make the filament back-off before a non printing move
#  on a FDM 3d printer, to avoid ooze.
my $retractLine = "G1 F$retractSpeed.* E";

# pattern of an unretraction line - would make the filament forward back to its position before starting extruding again.
my $unretractLine = "G1 E.* F$retractSpeed";

# read stdin and any/all files passed as parameters one line at a time
while (< > ) {
	if (/$retractLine/) {
		# if we found a retraction line, replace it with laser power off
		print "$powerOffLine";
	}
	else {
		# nothing to change, print line as-is
		print or die $!;
	}
	
	if (/$unretractLine/) {
		# if we have an un-retraction line, append laser power on right after it
		print "$powerOnLine";
	}
}

The link in the output of Slic3r is:
/Users/marinusdebeer/Documents/lasercuts/slicerscript_laser1.pl

I've ran the script many times with small changes each time, I really want the Gcode to be changed so I can work with my laser cutter, but somehow it just doesn't want to run it.

What am I doing wrong?
There are no errors, Slic3r reports the post-processing is successfully applied but nothing changes. I also tried printing out certain texts to see if it would output them into the GCODE, but even that didn't work so there must be something going wrong where Slic3r runs the perl script.

Hope anyone here can help me out, if we fix this, I can report back on the Reprap forums, it will make a few people that I know very happy! (Also some from the SeemeCNC forums).

- Marinus

alexrj, 2015-01-17:
Try to run the script as standalone:

```bash
/Users/marinusdebeer/Documents/lasercuts/slicerscript_laser1.pl /path/to/a/file.gcode
```

Maybe you haven't given execution permissions to it. In that case you need to do:

```bash
chmod +x /Users/marinusdebeer/Documents/lasercuts/slicerscript_laser1.pl
```

Ohmarinus, 2015-01-17:
Okay I will try that, does this mean the automatic option through Slic3r won't work? Thanks for your reply.

alexrj, 2015-01-17:
Slic3r just calls the script. I'm suggesting you try to do the same thing that Slic3r would do in order to see that it's not a Slic3r problem.

Ohmarinus, 2015-01-17:
Yes, I have tried the standalone way and now see that this also doesn't work. But I get no feedback in terminal if/why it doesn't work.

alexrj, 2015-01-17:
Then you see that it's not a Slic3r problem.

I suggested one thing in my answer above. Try that.

Ohmarinus, 2015-01-17:
You mean chmod +x the script? I did that right away. No change sadly.

Ohmarinus, 2015-01-17:
I have found out the problem through stackoverflow.

The solution is:
while (<> ) {

instead of:
while (< > ) {

closed 2015-01-18 by alexrj