slic3r/Slic3r #2558: perl Build.PL --gui fails on gentoo x86_64 with perl 5.20.1

swsch, 2015-01-17:
I'm trying to build Slic3r from git with GUI, but so far I only get this:

```
g128 Slic3r # git pull
Already up-to-date.
g128 Slic3r # perl Build.PL --gui
App::cpanminus is up to date. (1.7024)
Class::Accessor is up to date. (0.34)
Growl::GNTP is up to date. (0.20)
--> Working on OpenGL
Fetching http://www.cpan.org/authors/id/C/CH/CHM/OpenGL-0.6704.tar.gz ... OK
Configuring OpenGL-0.6704 ... OK
Building and testing OpenGL-0.6704 ... OK
Successfully reinstalled OpenGL-0.6704
1 distribution installed
--> Working on Wx
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Wx-0.9923.tar.gz ... OK
Configuring Wx-0.9923 ... N/A
! Configure failed for Wx-0.9923. See /root/.cpanm/work/1421529721.30449/build.log for details.
--> Working on Wx::GLCanvas
Fetching http://www.cpan.org/authors/id/M/MB/MBARBON/Wx-GLCanvas-0.09.tar.gz ... OK
==> Found dependencies: Wx::build::MakeMaker
--> Working on Wx::build::MakeMaker
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Wx-0.9923.tar.gz ... OK
Configuring Wx-0.9923 ... N/A
! Configure failed for Wx-0.9923. See /root/.cpanm/work/1421529725.30632/build.log for details.
! Installing the dependencies failed: Module 'Wx::build::MakeMaker' is not installed
! Bailing out the installation for Wx-GLCanvas-0.09.
Don't worry, this module is optional.
The following prerequisites failed to install: Wx
```
I've looked into the build.log for Wx:
```
...
Unpacking Wx-0.9923.tar.gz
Entering Wx-0.9923
Checking configure dependencies from META.json
Checking if you have Alien::wxWidgets 0.25 ... Yes (0.65)
Configuring Wx-0.9923
Running Makefile.PL
Option spec <> cannot be used with pass_through
Compilation failed in require at Makefile.PL line 21.
BEGIN failed--compilation aborted at Makefile.PL line 21.
-> N/A
-> FAIL Configure failed for Wx-0.9923. See /root/.cpanm/work/1421529721.30449/build.log for details.
```

Side note: Why does it run the OpenGL stuff every time? The other packages seem to be recorded.

alexrj, 2015-01-18:
I have no idea about that "Option spec <> cannot be used with pass_through" error. It's probably a bug in the Wx distribution, so you might want to report it there.
OpenGL is currently reinstalled every time as a workaround to a bug that requires the `--force` flag. I'll check whether there are lighter workarounds instead.

swsch, 2015-01-18:
Thanks for the reply. I'll try to dig into the Wx stuff, but my perl has gotten a bit rusty, so it will be interesting :-)

closed 2015-01-18 by swsch

dexy31337, 2015-01-18:
Have same here on Mac OS. @swsch if you'll found how to fix it, would you be so kind to share here?

swsch, 2015-01-19:
I tried the same thing as above on my laptop, which has 6 years less of stale cruft. Worked like a charm and slic3r-1.2.6-dev looks pretty :-)

Let's see if cleaning out /usr/local/lib64/perl5 does something on the old box.

swsch, 2015-01-19:
It does not ... so I'll have to compare stuff that's installed by the packaging system.

NewTheoryMan, 2015-01-23:
I'm having the same problem running Ubuntu 14.04 LTS 64bit. I have the same wx error when I try the --gui as you, @swsch. I am also unable to run the precompiled version. I can run Slic3r through Terminal, though.

kefir-, 2015-01-23:
@NewTheoryMan This is how I build on Ubuntu 14.04 LTS 64bit, try it out if you like. Note that I run Build.PL without sudo, and that's why I have to set the PERL5PATH. I prefer it this way, because then I can keep my system libs separate from slic3r's requirements. This way I can (in theory) also keep two completely different versions of Slic3r with completely different library requirements on the same system. I tried to install as many of the requirements (and requirements of requirements...) with apt-get, so that I get updates from Ubuntu.

    sudo apt-get install git build-essential # and xauth for ssh -X to headless/lxc
    sudo apt-get install cpanminus libencode-locale-perl libextutils-parsexs-perl perl-modules perl-base libmath-planepath-perl libmodule-build-withxspp-perl libmoo-perl libio-stringy-perl libclass-xsaccessor-perl libxml-sax-expatxs-perl libclass-accessor-perl libwx-perl libopengl-perl libglu1-mesa-dev mesa-common-dev libglu1-mesa-dev libdata-uuid-perl libcrypt-cbc-perl libmodule-build-tiny-perl libextutils-helpers-perl libio-socket-portstate-perl libextutils-installpaths-perl libmodule-runtime-perl libtest-fatal-perl librole-tiny-perl libtry-tiny-perl
    export PERL5LIB=$HOME/perl5:$HOME/perl5/lib/perl5 # add to .bashrc as well
    git clone https://github.com/alexrj/Slic3r.git
    cd Slic3r
    perl Build.PL # no sudo necessary, downloads are put in $HOME/perl5
    perl Build.PL --gui # same as above

swsch, 2015-01-23:
I compared the cpan module listing on both machines, they are very similar with differences only in stuff that should be totally unrelated to Slic3r.

@kefir- I tried your method, but I end up at the same problem as before ... Something weird is going on.

xxmicloxx, 2015-01-29:
Same problem here, on OS X using perlbrew... Does anyone know a solution?

mihai9, 2015-01-31:
i have the same problem
slackware 64 current perl 5, version 18, subversion 1 (v5.18.1)
googling i foud only  http://www.nntp.perl.org/group/perl.perl5.changes/2015/01/msg43698.html
from January 14, 2015 16:23

@@ -365,6 +373,9 @@ sub GetOptionsFromArray(@) {
 		next;
 	    }
 	    $linkage{'<>'} = shift (@optionlist);
+	    if ( $passthrough ) {
+		$error .= "Option spec <> cannot be used with pass_through\n";
+	    }
 	    next;
 	}
i am a total outsider to perl but maybe this helps

mihai9, 2015-01-31:
success (sort of)
looks that http://www.nntp.perl.org/group/perl.perl5.changes/2015/01/msg43698.html breackes wx (for the moment at least)
this is what i did (as regular user):

```
cpan App::cpanminus
CPAMN=~/perl5/bin/cpanm perl Build.PL
CPANM=~/perl5/bin/cpanm perl Build.PL --gui
```
edit ~/perl5/lib/perl5/Getopt/Long.pm and comment three lines above (376 to 378 for me)
```
CPANM=~/perl5/bin/cpanm perl Build.PL --gui
perl -I$HOME/perl5/lib/perl5  ./slic3r.pl
```
and it works
you have to run it as

```
perl -I$HOME/perl5/lib/perl5  ./slic3r.pl
```
or set some environment variables for that
this is certainly not the way to do it unless you can't wait
you could create a new user and try

swsch, 2015-02-01:
@mihai9 Removing these three lines allows Wx to install successfully on my previously non-working box. Thanks for discovering this.

I still have no idea why I didn't have to do this on my other box, where building the GUI worked fine on the first try.

Looks like Wx needs some updating to work with current systems, as Alien::Wx won't build with gcc-4.9.2 ... had to revert to 4.8.4 to get everything done. Sigh...

lordofhyphens, 2015-02-02:
There's some other wx3 related problems but those are all centered around
text boxes.

chron0, 2015-02-09:
Same trap happened on gentoo although portages Getopt doesn't have this patch included so it would have worked if Slic3r wouldn't bring its own cpan :) If you run into this issue on gentoo you can quick-hack it to get Slic3r gui working again, the "tainted" Long.pm was found in

    $ vi /usr/local/lib64/perl5/5.18.2/Getopt/Long.pm

Line 375FF, comment in the if clause up to the next:

            $linkage{'<>'} = shift (@optionlist);
            #if ( $passthrough ) {
            #       $error .= "Option spec <> cannot be used with pass_through\n";
            # }
            next;

then just

    $ perl Build.PL --gui

and it works again. I only wanted to simply update in order to have current Slic3r screenshots for our 3d printer documentation on https://apollo.open-resource.org/mission:resources:picoprint but instead it was time to hack around in perl deps... Felt like the 90's again though :)

Anyhow, be that is it may be for now, it's still great to have more than one open-source slicing toolkit available, so @alexrj, thanks for investing your lifetime into it and keeping it open and free. Much appreciated.