slic3r/Slic3r #2560: Thin walls still not filled?

bstott, 2015-01-18:
Thin walls are not filled in for my example. I've seen some Slic3r issue history but, I have the event in the recent versions 1.2+, which I use.

I have tried varying infill percentages, different fill patterns, varied loops/perimeters, different layer height, the detect thin walls option. My walls are approx. 1.5mm and should be easily filled. Right? 

I've attached a Slic3r 1.2.5 gcode result image which appears the same as other 1.2.x versions. I have also included a KISS result which was without any effort on my part to get that particular filled wall result.

 

[image]
[image]

alexrj, 2015-01-18:
Comparisons are not much helpful. I need files to reproduce the test cases (STL files and config).

zapta, 2015-01-19:
Same problem here.  Please see the test object here http://www.thingiverse.com/thing:642051  . Slic3r doesn't stitch the vertical layers which result very soft walls. Cura does it right which result in strong walls.

I tried to attach the generated gcode here but github wouldn't let me (file type error). Please load the STL from thingieverse and generate the gcode. Slic3r doesn't fill between the perimeters.

I noticed this problem when I tried to print this thing with slic3r http://www.thingiverse.com/thing:168437 , the printer worked all night and in the morning I had a beautiful object that was soft like paper because the vertical layers  were not stitched to each other.

Lord2Vader, 2015-01-20:
Same here. Raspberry Pi - Cases have thin walls and they are like foil

Mandlbaur, 2015-01-21:
https://github.com/alexrj/Slic3r/issues/2346#issuecomment-63932911

Vicious-one, 2015-01-25:
I've encountered this too. Sometimes can be "fixed" by adjusting specific infill width a little. Stitching would be good, though.

bstott, 2015-01-25:
Below is my config file. I have found the setting which will affect changes to the walls being filled or not - Print Settings > Layer and perimeters > Layer height. The issue with my latest project is not bound to the model. So, it appears any thin walled files exhibit the same issue as illustrated in my previous image files.

Replicating the hollow wall issue.:
Hollow walls - Set the Layer height to some smaller value - example: 0.24mm.
Filled walls - Set the Layer height to a value higher - example: 0.34mm.

P.S. In finding the trigger for this issue I've also found that some proper (?) *.stl files, published from Autodesk Inventor will crash Slic3r but, will work fine in MakerBot, Kiss, Netfabb, Meshlab and Cura. **How can I send you an *.stl file?** 

Here is my configuration file:
I have seen that linear infill wismall layer height of .24 will create a no filll two layer wall but, with a thicker layer height of .32 and infill of rectilinear

# generated by Slic3r 1.2.3 on Sun Jan 25 13:26:03 2015
avoid_crossing_perimeters = 0
bed_shape = 99.4522x10.4528,97.8148x20.7912,95.1057x30.9017,91.3545x40.6737,86.6025x50,80.9017x58.7785,74.3145x66.9131,66.9131x74.3145,58.7785x80.9017,50x86.6025,40.6737x91.3545,30.9017x95.1057,20.7912x97.8148,10.4528x99.4522,0x100,-10.4528x99.4522,-20.7912x97.8148,-30.9017x95.1057,-40.6737x91.3545,-50x86.6025,-58.7785x80.9017,-66.9131x74.3145,-74.3145x66.9131,-80.9017x58.7785,-86.6025x50,-91.3545x40.6737,-95.1057x30.9017,-97.8148x20.7912,-99.4522x10.4528,-100x0,-99.4522x-10.4528,-97.8148x-20.7912,-95.1057x-30.9017,-91.3545x-40.6737,-86.6025x-50,-80.9017x-58.7785,-74.3145x-66.9131,-66.9131x-74.3145,-58.7785x-80.9017,-50x-86.6025,-40.6737x-91.3545,-30.9017x-95.1057,-20.7912x-97.8148,-10.4528x-99.4522,0x-100,10.4528x-99.4522,20.7912x-97.8148,30.9017x-95.1057,40.6737x-91.3545,50x-86.6025,58.7785x-80.9017,66.9131x-74.3145,74.3145x-66.9131,80.9017x-58.7785,86.6025x-50,91.3545x-40.6737,95.1057x-30.9017,97.8148x-20.7912,99.4522x-10.4528,100x0
bed_temperature = 100
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 0
complete_objects = 0
cooling = 0
default_acceleration = 0
disable_fan_first_layers = 0
dont_support_bridges = 0
duplicate_distance = 6
end_gcode = G92 E0 ;         Reset Extruder to Absolute 0.\nG1 E-5 F800 ;    Retract filament 5mm.\nG28       ;      Home all axes.\nG92 E0 ;         Reset Extruder to 0. \nM140 S0   ;      Set Bed temp. to 0.\nM104 S0   ;      Set Hot End temp. to 0.\nM106 S255 ;      Fan 100% to cool Hot end.\n;                Print tune on Completion:\n;                'Mission Impossible.'\n\nM300 S3135 P150\nM300 S0 P300\nM300 S3135 P150\nM300 S0 P300\nM300 S1864 P150\nM300 S0 P150\nM300 S4186 P150\nM300 S0 P150\nM300 S3135 P150\nM300 S0 P300\nM300 S3135 P150\nM300 S0 P300\n\n;M300 S1864 P150\n;M300 S3135 P150\n;M300 S2349 P1200\n;M300 S0 P75\n;M300 S1864 P150\n;M300 S3135 P150\n;M300 S2217 P1200\n;M300 S0 P75\n;M300 S1864 P150\n;M300 S3135 P150\n;M300 S2093 P1200\n;M300 S0 P150\n;M300 S932 P150\n;M300 S2093 P150\n;            End of final print tune.\n\nG4 S65;      Pause/Dwell 65secs to cool Hot End.\nM107  ;      Turn Fan off.\nM18   ;      Disable stepper motors.\nM300 S3135 P150\nM300 S0 P300\nM300 S3135 P150\nM300 S0 P300\n\nM190 R65 ;          Bed Cooling. Wait for temp.\nM140 S0   ;          Set Bed temp. to 0.\nM300 S1864 P150\nM300 S3135 P150\nM300 S2349 P1200\nM300 S0 P75\nM300 S1864 P150\nM300 S3135 P150\nM300 S2217 P1200\nM300 S0 P75\nM300 S1864 P150\nM300 S3135 P150\nM300 S2093 P1200\nM300 S0 P150\nM300 S932 P150\nM300 S2093 P150
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 80%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 0.9
extrusion_width = 0
fan_always_on = 1
fan_below_layer_time = 60
filament_diameter = 1.75
fill_angle = 56
fill_density = 30%
fill_pattern = line
first_layer_acceleration = 0
first_layer_bed_temperature = 100
first_layer_extrusion_width = 120%
first_layer_height = 0.34
first_layer_speed = 40%
first_layer_temperature = 220
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 70%
infill_first = 0
infill_only_where_needed = 0
infill_speed = 50
interface_shells = 0
layer_gcode = 
layer_height = 0.24
max_fan_speed = 100
min_fan_speed = 70
min_print_speed = 10
min_skirt_length = 5
notes = 
[... 40 lines cut ...]
support_material_enforce_layers = 1
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 2
support_material_interface_spacing = 0.2
support_material_interface_speed = 100%
support_material_pattern = rectilinear-grid
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 0
temperature = 220
thin_walls = 0
threads = 4
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 40
top_solid_layers = 3
travel_speed = 180
use_firmware_retraction = 0
use_relative_e_distances = 0
vibration_limit = 0
wipe = 0
xy_size_compensation = 0
z_offset = 0.4

GHPS, 2015-01-25:
I can confirm with Slicer 1.2.5-dev that the "Detect thin walls" option actually creates hollow walls. With the new (fantastic) preview I could finally track down a problem I had for months (and verious versions of Slicer) without any clue where the real cause of the problem was to find (infill, perimeter, shells). Take for example this file http://www.thingiverse.com/download:503695.

With "Detect thin walls" turned on the file is printed with an open top surface:
[image]

Turn off the thin walls option and everything is fine:
[image]

dexy31337, 2015-01-25:
Had same problem in https://github.com/alexrj/Slic3r/issues/2515

bstott, 2015-01-25:
The hollow and filled thin walls can be created by having the 'detect thin walls' turned off and adjusting the layer height. The 'detect thin walls' on and off changes the printing of some part's tops. So, it seems that the two settings of; 'detect thin walls' and 'Layer height' have an effect with generating hollow and filled features (walls and part tops). Yes?

bstott, 2015-01-27:
The change point where thin walls are not filled in --- layer height: <=0.3mm

Vicious-one, 2015-01-27:
https://github.com/bstott -- Are you using "automatic" path width (calculated from layer height) or define them manually?

bstott, 2015-01-27:
@Vicious-one -- I am not familiar with "automatic" path width. So, it is likely what is set and I am using it.

I would think the width is a function of the volume and height thus not something an operator would need to fight with too?

Vicious-one, 2015-01-27:
@bstott -- Well, actually, an operator will be forced into it either way :-)

Width is determined by matching print head and extrusion linear speed, then applying certain limits, so yes, it is [loosely] connected to layer hight. But you can override that using Advanced tav in Printer settings.

I usually tweak perimeter widths a little until gaps become mostly filled, but triple thin wall is not created.

The problem is, gap fill flow is calculated in a strange way and is either absent, too thin or too abundant before tweaking.

bstott, 2015-01-27:
@Vicious-one 

OK, I can see what you are driving at - But, Slic3r is not generating the fill paths between thin walls when layer height is below a threshold value, in this case. I see the 'Default extrusion width' setting you've referred to in Slic3r's 'Print Settings>Advanced'. 

Maybe this is the problem with some infill not being placed - When the layer height is small and constant plastic volume the width of the extruded plastic will be calculated to be wider and thus infill is not generated by Slic3r. If this is how the hollow walls are occurring then the application's present calculations require some refinement to still place fill without our having to adjust other settings. We should not need to go advanced and tweak the 'Default extrusion width' under 'Print Settings>Advanced'. 

In my specific part to print the walls are designed 1.5mm in width. The printer nozzle is 0.5mm. I must set a layer height above 0.3mm to fill the wall, otherwise, Slic3r generates tool paths for a single width exterior, single width interior and a hollow core. In this case, the hollow wall space appears as large or larger than another single width extruded thread of plastic.

I've finished printing this last project but, will try to remember the 'Default extrusion width' setting for next time. Thanks!

Lord2Vader, 2015-01-28:
Hi,
I have the Default extrusion width setting (0) and all but top solid infill set to 0.
Even some 2mm thick walls are not filled correctly with a 0,4mm nozzle

Lord2Vader, 2015-01-29:
Hi,

ive testet it again. Every extrutionwidth set to 0.
@0.4mm Nozzle i get an infill at 0.25mm layerheight
@0.5mm Nozzle i get an infill at 0.30mm layerheight

So there has to be a little calc error? Cause in theory at 0.1mm layerheight it should be easier to infill that than in bigger ones cause of the roundness of the layed out line?

Would be nice of you could fix that. At 0.25 mm layerheight small things dont look as good but in 0.1mm layerheight you get no infill causing as said above Pi-Cases are week as sh*t.

To me it seems like a real bug!

alexrj, 2015-02-15:
@bstott, you need to host your STL file somewhere and put a link here. Without a STL file, I can take no action.

Also, I recommend to use the new 3D preview in Slic3r (1.2.6+) for taking screenshots, as it shows also the thickness of the extrudates. Any preview which only displays the toolpath centerlines it not useful as it doesn't show whether there will be gaps or not.

alexrj, 2015-02-15:
I pushed a change that makes gap fill more aggressive. @zapta, this is how your file looks like now:

[image]

alexrj, 2015-02-15:
@Lord2Vader, please open a separate issue and provide the things listed in the guidelines, otherwise no action can be taken on your report.

lordofhyphens, 2015-02-17:
It looks to me like the problem has been sorted out. 
STL:  https://www.dropbox.com/s/cz073mutkiu23vz/t400-back-0.3.STL?dl=0
1.2.6-release: https://dl.dropboxusercontent.com/u/1360727/Screen%20Shot%202015-02-17%20at%2012.28.24%20PM.png
1.2.6-dev: http://i.imgur.com/Fs0E2AJ.png

Wnt, 2015-02-19:
I'm still experiencing this problem or a similar problem when using automatic extrusion widths. I'm using Slic3r 1.2.7-dev built from d44bf389068e57960ec1239bffe4ea5ba4d9c7df

[image]
No propper infill when using automatic extrusion widths

[image]
I can get propper infill by manually setting the extrusion widths

config bundle:
http://upload.egarden.fi/Slic3r_config_bundle_infill_problem.ini

NOT OK settings from the bundle:
Print: slower
Filament: hipst_test_1
Printer: Untitled

OK settings from the bundle:
Print: spool_base
Filament: hipst_test_1
Printer: Untitled

The detect thin walls feature is off for both print settings. In the automatic extrusion width (the not OK) version switching the detect thin walls feature on doesn't fix the hollow thin walls problem. In the manual extrusion width (the OK) version the detect thin walls feature uses less "zigzag" paths in the infill.

Test file:
http://upload.egarden.fi/thin_wall_test.stl

NOT OK G-code
http://upload.egarden.fi/thin_wall_test_NOT_OK.gcode

OK G-code
http://upload.egarden.fi/thin_wall_test_OK.gcode

alexrj, 2015-05-24:
Thank you @Wnt, that was an edge case caused by your layer_height = 0.15. I fixed it:

[image]

closed 2015-05-24 by alexrj