slic3r/Slic3r #2561: v1.2.5 Strange artifacts

justmyopinion, 2015-01-18:
Some strange artifacts at the limits of this model
[image]

And specific strange layout of layer at 1mm:
[image]

STL file:
https://www.dropbox.com/s/jz34hlps8fsaqre/Idler_GT2_2mm_lip.STL?dl=0
Config
https://www.dropbox.com/s/ptowwqw8tw3e0n1/artif_config.ini?dl=0

alexrj, 2015-01-18:
The artifacts in the first screenshot are just rendering errors, don't trust them. They are acute angles not being mitered.

The artifact in the second one is an issue of the Clipper library. I'll investigate it.  Same thing as #2416

alexrj, 2015-01-30:
Well, the reason for the artifact is that the original STL file has a little spike and that's being propagated in the offsets:

[image]

So it's theoretically correct. But the result is not visually nice and gives the user a feeling that Slic3r is misbehaving, so I'll check whether this is fixable without slowing things down.

alexrj, 2015-01-30:
Okay, I added a call to the Douglas-Peucker simplification algorithm before generating perimeters. Small performance penalty, sadly.

closed 2015-01-30 by alexrj

a4jp-com, 2015-01-30:
It would be very nice if this could get fixed. Even with a speed penalty it is nicer to see a clean image. At the moment it just looks like the program is playing up.