slic3r/Slic3r #2563: Feature request for Plater: Show brim/raft in plater?

lordofhyphens, 2015-01-19:
Not a huge problem, but it'd be very convenient to be able to see the configured brim (or a raft) on the 2d (or possibly 3d plater if that's a plan), similar how the skirt is shown currently.

a4jp-com, 2015-01-19:
I would like the same if possible ^^

lordofhyphens, 2016-07-11:
It is there noe for the 3d plater

a4jp-com, 2016-07-12:
I just updated but I couldn't see a change on the 3d platter. The 2D platter has a drag and refresh bug though.

lordofhyphens, 2016-07-12:
Right, I was thinking about the 3d preview.

bubnikv, 2016-07-13:
I fixed the drag and refresh bug on the 2d platter on Windows in the
prusa3d fork. You may verify it with the prusa3d build.

<https://github.com/alexrj/Slic3r/issues/2563#issuecomment-232211259>, or mute
the thread

lordofhyphens, 2016-07-13:
What commit was it?

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-07-13:
prusa3d:ef0d22be3047b7a77e65ba4a76ef5b274daa06c7 ?

lordofhyphens, 2016-07-13:
nvm, I think I found it; PR is #3416

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-07-13:
@a4jp-com was that bug ever reported to Slic3r? My search-fu isn't bringing it up.

a4jp-com, 2016-07-13:
I think it was but I can make a new post if doesn't exist.