slic3r/Slic3r #2565: Solid infill on too many layers

fpiesik, 2015-01-19:
Slic3r Version: 1.2.5
OS: Ubuntu 14.04 64Bit

expected result: the solid infill should be only at the top layers
actual result: solid infill is generated on too many layers on the left and righ side of the model.

This is problably because the sides are rounded and there are effectively no top layers:
[image]

here is a pictiure of the g-code at less than half heigt, note the solid infil left and right:
[image]

https://www.dropbox.com/sh/sp2nr2o58ojco7b/AAC0JzIdaJLckyfvlqpEIsJxa?dl=0

justmyopinion, 2015-01-19:
Try to adjust your Solid Infill Treshold:  at Printing/Infill/Solid......

fpiesik, 2015-01-19:
that helped - thanks!

closed 2015-01-19 by fpiesik