slic3r/Slic3r #2566: Rectilinear support material not correctly lined up

Gatulibu, 2015-01-19:
In recent versions the support material isn't generated correctly:

Version 1.2.5:
[image]

Version 1.1.7 generates correct support:
[image]

example STL, config files, gcode:
https://drive.google.com/folderview?id=0B2EPQumqM4OhN2dobkNnYUFHd00&usp=sharing

alexrj, 2015-01-25:
Interesting, thank you for that report.

simonkuehling, 2015-01-25:
I was just writing a new issue on this - but seems like this one covers my findings already.
I'd like to add that the misalignement affects rectilinear as well as rectilinear grid support structures
[image]

alexrj, 2015-01-28:
Fixed, thank you!

[image]

closed 2015-01-28 by alexrj