slic3r/Slic3r #2568: Request: Have selected printer setting remain when switching between saved printer settings

griffdb, 2015-01-20:
In Slic3r 1.1.7, when in expert mode, if you select Print or Filament settings, and then select one of the sub menus, such as "Skirt and Brim" or "Cooling", if you switch between the saved settings for that category, the selected sub menu remains, allowing for easy comparision between saved settings.   

By contrast, if you select printer settings, and you select the sub menu "Custom G-code", if you load another saved setting, then the selected menu goes back to "General", making comparisons difficult between the two saved settings of "Custom G-code" or "Extruder 1".  Please implement the loading of saved settings in Printer settings to have the behavior as that of Print and Filament settings.
[image]
[image]
[image]
[image]

griffdb, 2015-01-20:
Also, when saving printer settings, please allow the current submenu to remain active.  Currently, when you save the printer settings, the sub menu gets reset to "General".  Please have the behavior be similar to when the Print and Filament settings are saved.  The active submenu remains active.

alexrj, 2015-01-25:
Good catch, I fixed that.

closed 2015-01-25 by alexrj