slic3r/Slic3r #2569: [v1.2.6-dev] crashes with "std::bad_alloc" after loading STL file

simonkuehling, 2015-01-20:
Hi @alexrj,

just a quick crash report for the current 1.2.6-dev @ c9cdae1a9659496f06066eb612e2baf55edfe4c4 - Slic3r exits repeatably when loading the attached simple STL onto the plater after a fresh start (but is working quite nicely with other models...):

```
$ ./slic3r.pl 

(slic3r.pl:20806): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 24

(slic3r.pl:20806): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 24

(slic3r.pl:20806): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 24
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Abgebrochen (Speicherabzug geschrieben)
```

Ubuntu 14.04 64bit

[STL file (sphere)](https://www.dropbox.com/s/4l18w9kb3c3wbi5/dual%20extrusion%20-%20sphere%20on%20support%20-%20sphere.stl?dl=1)
[config](https://www.dropbox.com/s/hnuy7g19a6emju9/config.ini?dl=1)

justmyopinion, 2015-01-20:
When I load your config file i get this warning from Slic3R, may be some limitation to this version:
[image]

You have selected support and new param: distance to object=0, which seem has some issues and makes slic3R unstable, but I have no more info on that, other than what is reported at this site.
[image]

If this param is different than 0 it seems to slice normally.

simonkuehling, 2015-01-20:
hi @justmyopinion - thanks for verifying my report!
the config file is actually exported from the selected presets currently used (for easier debugging), but you're right - this does not work that well with mutli-extruder configurations.

Anyway: i can confirm that it is caused by the just recently introduced "soluble" z-distance support option (=0) - and changed the issue title accordingly

Vicious-one, 2015-01-20:
Seems the same: https://github.com/alexrj/Slic3r/issues/2567

alexrj, 2015-01-25:
I can only reproduce this by opening the 3D preview tab. If I don't do that, I get no crash even when exporting the G-code. Can you confirm it's the same for you?

Vicious-one, 2015-01-25:
I'm getting crashes even if I do not open any tab except for 2d plater.

simonkuehling, 2015-01-25:
Like @Vicious-one this also happens without opening the 3D preview - fresh start, 2d plater, add part, no further interaction (but background processing activated, of course)...

Anything else we can deliver for debugging?

alexrj, 2015-01-25:
Weird, I got that error with that model and those settings only in the 3D preview rendering code. I fixed it now. I don't understand why you got that error without opening 3D preview...

Can you git pull and rebuild now? I wonder if that's now fixed...

Vicious-one, 2015-01-25:
I confirm the crash does not occur anymore for me.

gege2b, 2015-01-25:
hey, it seems to fix #2579 as well !

simonkuehling, 2015-01-25:
Yes, it is working now! Great.

closed 2015-01-26 by alexrj