slic3r/Slic3r #2571: Feature Request: Split up X/Y compensation so can have different compensations

lordofhyphens, 2015-01-21:
The X/Y compensation is neat, but can we get it broken out into the X and Y components?

whosawhatsis, 2015-01-21:
You're missing the point. This is not meant to compensate for mis-calibrated X/Y steps/mm settings, it's designed to compensate for the tolerance of the surfaces due to the way the plastic behaves after leaving the extruder.

alexrj, 2015-01-21:
@whosawhatsis is totally right.

And you know that I hate starting from solutions without having even discussed the problems. Any request or proposal should be introduced by a good reason, otherwise the answer to your question would just be "yes, you could but I won't" :-P  I hope you'll understand.

If your purpose is to fight backlash, that's a mechanical problem which must be handled physically. There was a firmware command for backlash compensation (`M99`), since mechanical properties are better handled by the firmware, but it was removed from Marlin.

alexrj, 2015-01-21:
(By the way, XY compensation is not the same thing as backlash compensation because it uses a very very different algorithm that I cannot really split into X/Y. Pure backlash compensation can be achieved with a simple G-code post-processing script)

closed 2015-01-21 by alexrj

lordofhyphens, 2015-01-21:
Thanks for the info, I'll go a-digging then.