slic3r/Slic3r #2572: 1.2.6-dev: Slicing provided AMF causes Slic3r to consume 6+GB of RAM

lordofhyphens, 2015-01-21:
STL and config: http://www.2shared.com/file/nOBaE9Zw/memory_consume.html (second download button)

What it says in the title. Every time I try to export Gcode for the provided STL, Slic3r's memory usage balloons to 90% of my available system RAM, which freezes up my UI and every other program (due to disk swap).

The 5mm step cube is fine though.

Wnt, 2015-01-21:
I can confirm the memory consimption problem. Just built Slic3r from the latest master (8cfd2e33d87701c3ccf947e89ef4dc146e30d7f3) and loading this file
http://upload.egarden.fi/All_parts_MakerBot_Replicator_2_and_2X.stl
causes Slic3r to consume all available memory after the status bar goes blank again after going through "Generating perimeters...", "Preparing infill..." and "Infilling layers...":
[image]
here is the --datadir i used:
http://upload.egarden.fi/slic3r_datadir.tar.bz2
I'm running on x86_64 Ubuntu 14.04, Perl 5.18.2

Wnt, 2015-01-21:
Like @gege2b commented on the f2818dd commit, the new Live 3D toolpaths preview feature seems to cause huge memory consumption. Reverting to 3648bbd6d4060c71500a241cb7214aab178202a8 fixes the memory consumption problem.

c-born, 2015-01-21:
Confirmed on Win7_64, 12GB Ram.
Build 8cfd2e33d87701c3ccf947e89ef4dc146e30d7f3 3D Toolpath preview bogged the system down with paging, took an hour before I got control back. If I load and don't go to the toolpath preview I can export Gcode without a problem.

lordofhyphens, 2015-01-23:
I had the 3D view open, but never touched the 3D toolpath preview.

alexrj, 2015-01-25:
This should be fixed now.  I still need to check why it seems that 3D toolpath rendering is taking place also for some users that don't open its tab. Maybe it's a platform-dependent behavior of wxWidgets.

lordofhyphens, 2015-01-25:
I have both wx 2.8 and 3.0 installed, but I don't know offhand which slic3r is using. 

3.0.2-1+b1
2.8.12.1+dfsg2-2

lordofhyphens, 2015-01-25:
It looks like it's fixed, at least for that test case.

closed 2015-06-02 by alexrj