slic3r/Slic3r #2573: Crash on export g-code (Segmentation fault)

xix-xeaon, 2015-01-21:
When I try to export this model to g-code Slic3r crashes with `Segmentation fault (core dumped)`.
I used 1.2.5 from website on Linux x64.
STL: https://dl.dropboxusercontent.com/u/53934565/phone_cover.stl
Config: https://dl.dropboxusercontent.com/u/53934565/slic3r_config2.ini

It works in the latest git through command-line (I'm not able to build the GUI):
```./slic3r.pl --load ../../Dropbox/Public/slic3r_config2.ini ../../Dropbox/Public/phone_cover.stl```

andbad, 2015-01-21:
Mee too.
STL: https://dl.dropboxusercontent.com/u/12390846/keychain-customizable-lithopane-mark1-4_20150120-11924-7s18kc-0.stl
Config: https://dl.dropboxusercontent.com/u/12390846/config.ini

 Same file/configuration, with 1.2.4b it's ok, instead.

By(t)e

GHPS, 2015-01-25:
I can confirm this bug for Slicer 1.2.5-dev under Ubuntu 14.04 LTS with this file from Thingiverse
http://www.thingiverse.com/download:503698. Trying to export the GCode caused Slicer to crash with a "perl[15579]: segfault at 0 ip b6fe00dd sp b0502f20 error 4 in XS.so[b6df0000+2e9000]" after exporting about 1.4 MB of the code. The last lines in the file are 

G1 X128.962 Y132.950 E1.45574
G1 E1.23074 F1500.00000
G92 E0
G1 X132.600 Y105.350 F9000.000
G1 X1

No change in the settings helped - I had to resort to Slicer  1.1.7 under Windows to print the file.

TTN-, 2015-01-26:
@andbad I've compile Slic3r with the the latest commit:  a5a0b672642e2490a86fd7e1b08cee3b7f3a9b4
Using your configuration and stl file and exporting gcode doesn't cause slic3r to crash. It appears to be fixed. Ubuntu 14.04 64bit,  Perl v5.18.2

alexrj, 2015-01-26:
Thank you, according to reports from @xix-xeaon and @TTN- this is fixed as of current git HEAD.

closed 2015-01-26 by alexrj