slic3r/Slic3r #2574: 1.2.5 Bed Shape not in Simple mode and not kept in Expert on Windows7 64

braddo99, 2015-01-21:
Starting with a clean installation (i.e. deleting User/Me/AppData/Roaming folder containing the ini files) when I open 1.2.5 the preferences wizard starts up - an option is given to choose bed shape and, in the case of Circular, diameter. (I guess the origin would always be 0,0 because no option given)

If you look at printer prefs, there is no way to see the bed shape. I think this is not an "expert" setting and should probably seen in simple mode. (I'm not sure it's actually set because of the below and also when I tried printing from here the effector seemed to attempt to go to the corner of a square bed, crashing the bot)

Once this is set from the wizard, if you then proceed to preferences and switch to expert prefs, restart Slic3r then check the Printer Set bed shape it has reverted to square.

alexrj, 2015-01-21:
Yeah. Simple Mode needs some love. If you have other ideas for it, tell me...

alexrj, 2015-01-25:
Okay, I added Bed Shape to Simple Mode too. Thank you!

closed 2015-01-25 by alexrj

braddo99, 2015-01-29:
Great. Will remember to jot down other improvements for simple mode as I come across them.