slic3r/Slic3r #2575: Feature Request: Add option to include support material in exported STL file

Kopfgeldjaeger, 2015-01-22:
Sadly, not all printers are supported by Slic3r. Some, especially industrial ones, require a proprietary slicer, but of course still accept STL files.

It would be great if there was an option to generate support material (and possibly infill changes, skirts, ...) and include those in an exported STL file. Of course, I see that many Slic3r options (temperature changes, etc.) are not possible to include in an STL file.

alexrj, 2015-01-22:
It's unclear to me how would you represent infill and skirt in a STL file… An STL file does not represent toolpaths, it just represents abstract volumes. Support material could be represented as a monolithic block of material below the object, but I don't think that would perform much better than each printer's specific software that can work at the toolpath level.

Can you provide more details and discussion?

Kopfgeldjaeger, 2015-01-22:
Yes, my idea was that support material is then represented as material below the object. This is of course not ideal, but (in my case) better than alternatives. My printer supplier does not even offer support generation in the slicing software...
I have already printed with support structures created by Meshmixer (and exported to STL), but they are not as good as Slic3r's as far as I can tell.

About infil/skirt I am not as sure, but I imagined that you can e.g. create the 3D geometry which is the infill and export that to volumes in STL (that idea only came when writing the feature request). I don't know how well that would work, and it's not as important to me as support structures.