slic3r/Slic3r #2578: [1.2.6-dev] UI locked while generating 3D Preview

gege2b, 2015-01-23:
Hi

Slic3r 1.2.6-dev 
Debian Jessie x64

Config not really important, model big/complex enough to take some time for the 3D Preview to be generated

As soon as we change a setting (say infill % for example), the UI became unresponsive while the 3D preview is generated. For a big/complex model, this can be very long.

If we have further setting to change, we must wait for the UI to be responding, change another setting, and waiting again, and redo this for the next settings...

alexrj, 2015-01-25:
True, that happens when settings are changed while the preview tab is open. We'll need to do that in a background thread...

gege2b, 2015-01-25:
the current head seems to solve this issue through, its OK now (I'll let you close this issue if nobody else complain about this)

alexrj, 2015-01-25:
Well, I think it's just faster so it's less noticeable... but still there are some lockups, so I think switching to a background thread is the correct solution anyway.

gege2b, 2015-01-25:
Its clearly less noticeable then 
I agree about the background thread anyway

gege2b, 2015-02-02:
just tested with 1.2.6-exp on win7 x64 and it works fine too

closed 2015-05-04 by alexrj