slic3r/Slic3r #2579: [1.2.6-dev] 3D Preview did not release memory

gege2b, 2015-01-23:
Hi, again
Same config as my precedent issue : Slic3r 1.2.6-dev on Debian Jessie x64
Again, this is about the 3D Preview, which seems to eat memory and never release it

Way to reproduce : open a model (not important which one), change settings (again, not important)
For each setting changed, I looked at the memory used by Slic3r using kSysGuard

Just opened, Slic3r use ~ 70MB of RAM
After a model as been imported (with no setting change) it moves up to 1.9GB
Changing layer height from o.3 to 0.2, he jumped to 2.6GB
Changing layer height back to 0.3 => 2.8GB
Changing infill % from 40 to 15 => 2.9GB

As you see, memory didn't seems to be correctly released at each step

an unknown problem occurs when I changed infill% back to 40%, Slic3r suddently eats up all available memory (~7.2GB) and my swap file grown from 0 to 5GB+ until I'd (painfully) terminated slic3r for my PC to be usable again

Its not the first time I get this with 1.2.6 using 3D Preview

A quick workaround would be to process and show  3D Preview only when the user want it (i.e by clicking on a new "3D Preview" button)

Anyway, making this great feature less hungry is a better solution :)

alexrj, 2015-01-23:
Yes, but it's not 3D preview. There's a leak in thread handling, regardless of 3D. I've been working on that for days.

gege2b, 2015-01-23:
Yeah, it make sense

I was suggesting 3D preview was faulty because I got this problem only since this feature appeared

alexrj, 2015-01-25:
Okay, now 3D preview consumes much much less memory than before.  Can you enjoy.. er, can you confirm?

gege2b, 2015-01-25:
It's building right now, you'll get my feedback soon

anyway, there is a missing package : Try::Tiny, so the "Moo" package cannot build and raise an error. Manually installing Try::Tiny solve the problem

gege2b, 2015-01-25:
After some test, it seems far better...... until Slic3r decides to eat again all the available memory

It happens as soon as I load one of this STL (and more randomly with other STLs) and with default config : 
https://dl.dropboxusercontent.com/u/34934879/formation%203D.STL
https://dl.dropboxusercontent.com/u/34934879/24hLANx6.STL

so I don't know if this is really an issue concerning the 3D Preview

gege2b, 2015-01-25:
just tested with the latest commits and the two STLs mentioned above, and slic3r is no longer hungry

closed 2015-01-28 by alexrj