slic3r/Slic3r #2580: Feature request: Optional filament temperature-profile

kyrreaa, 2015-01-23:
Having the ability to define a profile for the filament that relates flow-rate to temperature would deliver much better prints in situations where you have large objects with fine details and slow areas.
For example objects that taper to points etc would print fine at high temp and speed during the bulkier stages but require a slower speed to allow plastic to cool at the tip.
Temperature required to push a smooth flow at low speed is also much lower than for high speed so such a profile would be much easier than editing or post processing in order to tweak temp vs flow/feed.

As an example I always test my new plastic by extruding a larger volume continously while monitoring speed at multiple temperatures.  I find the level where my extruder's grip starts to slip and set that as my absolute max after reducing by approx 10%. Plotting the data I can see a very clear curve emerge.
[image]

A simple approximation would probably suffice and optionally a time-delay could be specified as (degrees c)/(minute) if needed to allow enough time to reach new temperature. Alternatively it could just transition naturally.

I am not talking about implementing instant temp transitions between infill and perimiter etc, I am talking about the average flow per layer as it changes during long print jobs.

alexrj, 2015-01-25:
So, you're talking about a relationship between temperature and **maximum** flow rate. If I get it correctly, you propose to adjust temperature according to the average layer printing speed?

kyrreaa, 2015-01-25:
Yes, for example.
Printing fine detail with low temp on the final layers of a design for instance, will not be compatible with a large bulk mass that is printed on the layers below.  Manual work with the code to adjust this based on layers is tedious and not for the novice user, even if it's a novice user of a professionally set up printer.
By letting the slicer know more about how the filament reacts to different temperatures it can easily adapt the temperature to the task at hand, preventing stripping of filament in the feeder and delivering cleaner fine details. There is already code to control the cooling based on layer time, so it should be easy to connect the dots and have some sort of temperature dependency to the flowrate or layertime.

kyrreaa, 2015-02-27:
Turns out Marlin firmware has had this (sort of) for a while (atleast from April 2013):
https://github.com/ErikZalm/Marlin#autotemp

I think letting slicer plan it would be better as it can look ahead much further than a printer can.  Especially on complex/detailed code.
It is also easier to interface with the PC for editing possible thermal curves etc.

pyrophreek, 2015-02-27:
This is an interesting idea, and I really like the graphs you  made.  It makes me wonder if Slic3r could be used to generate such a graph?  have an option that runs a little app that lets you test say three or four different temperatures with short pulses of extrusion, followed by a popup asking if the extruder slipped. Since Slic3r isn't designed to be a Gcode sender, I can understand that this might not be feasible, but perhaps a separate program that someone else writes could be integrated as an integral app to generate the data that Slic3r could use.

kyrreaa, 2015-02-27:
I am working on a new sensor for Rostock firmware that will make the feed closed loop. With the sensor in place it would be easy to let the firmware support automatic measurements of plastic flow capability.
There is a very interesting article written by "illuminarti" here:
http://www.extrudable.me/2013/04/18/exploring-extrusion-variability-and-limits/
It shows just how large a difference there could be depending on extrusion rate.

My own design has a variance of almost 10% within the interval I operate it, and since my filament diameter and actual machine motion is so precise this variance is the major contributor to surface issues. Printing identical objects on different parts of the build platform gives the same wall variation due to the head pressure and temp. To combat this, the closed loop control will make sure this slippage is compensated for, but to get fine details and good surface quality you also want to have your temperature better fit the object area. This is where people start to slice with multiple profiles according to layer height and such complex behavior when we could instead couple the temperature to the already calculated flow and layer time system in Slic3r.

Since most of the values needed are already calculated it could be interesting to see what happens if we just add a temp delay value so Slicer can know how fast the hot-end reacts and let it adapt the temperature either by layer or even fully live during a layer.
To take it a step further, planning small vs large perimiters and grouping them according to ideal temperature could improve things even more. This may be taking it too far though, and possibly it is best to keep it simple. The possible benefits of the initial step may be all we need.

VanessaE, 2016-06-11:
What I would like to see is a simple temperature modifier attached to the "Print" settings.  I.e. if you have a filament profile specifying a temperature of 190, you could set up a high-speed printing profile with a temperature modifier of say 15.  This would raise the filament temperature to 205 degrees for just that print profile.  Seems like that would be better than storing a bunch of filament profiles for different speeds (especially for folks who have dozens of different filaments at their disposal).