slic3r/Slic3r #2581: Slic3r 1.2.5 : Support bridge pattern angle ne respecte pas les valeurs indiquées.

bharralure49, 2015-01-23:
J’ai déjà essayé par deux fois d’aborder ce sujet, mais sans succès… Certainement que ma prose n’est pas de bonne qualité, mais je ne suis pas un homme de lettres, mais de terrain, perdu au fin fond de la France avec tous ses problèmes dus à la barrière de la langue.
Mais ce disfonctionnement étant à mon sens d’importance, et comme de plus je suis têtu, je renouvelle ma tentative…
Je persiste donc à dire que Slic3r ne respecte pas les angles d’orientation du remplissage sous les ponts, sauf pour la valeur par défaut : 0°.
Je ne fais pas de commentaires, mais je joins simplement des captures d’écran qui, à mon sens, parlent plus que de longs discours.
Pour terminer, je tiens à faire remarquer que la première couche de remplissage est toujours en décalage (sur le plan X Y) par rapport aux suivantes…

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

D’autre part, chacun sait qu’il est possible d’imprimer des ponts sans soutien jusqu’à une certaine longueur.
Actuellement Slic3r ne semble pas faire la différence entre petits et grands ponts.
Ne serait-il pas judicieux qu’un nouveau paramètre nous permette d’indiquer une valeur (minima ou maxima), et que chacun pourrait incrémenter ou décrémenter selon ses besoins ?

oschonrock, 2015-01-29:
I can confirm this on 1.25

lordofhyphens, 2017-03-19:
Can someone please check this on 1.3.0-dev? 

http://dl.slic3r.org/dev