slic3r/Slic3r #2583: 1.1.7 & 1.2.5 don't generate support where needed...

nothinman, 2015-01-24:
I've tried latest stable, and latest experimental, and slic3r doesn't generate support for the top of the object, only supporting area around the holes.

Complete object:

[image]

but lacks support:

[image]

My config:
```
# generated by Slic3r 0.9.10b on Sat Oct  5 18:11:54 2013
avoid_crossing_perimeters = 0
bed_size = 320,280
bed_temperature = 0
bottom_solid_layers = 5
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 150
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 3000
disable_fan_first_layers = 1
duplicate = 1
duplicate_distance = 6
duplicate_grid = 1,1
end_gcode = M104 S0 ; turn off temperature\nG28 X0  ; home X axis\nM84     ; disable motors
external_perimeter_speed = 50
external_perimeters_first = 
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0,0x0
extrusion_axis = E
extrusion_multiplier = 1,1
extrusion_width = 100%
fan_always_on = 
fan_below_layer_time = 60
filament_diameter = 2.90,3
fill_angle = 45
fill_density = 0.4
fill_pattern = rectilinear
first_layer_bed_temperature = 0
first_layer_extrusion_width = 0
first_layer_height = 110%
first_layer_speed = 30%
first_layer_temperature = 200,200
g0 = 0
gap_fill_speed = 20
gcode_arcs = 1
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 3000
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_speed = 120
layer_gcode = 
layer_height = 0.3
max_fan_speed = 100
min_fan_speed = 30
min_print_speed = 10
min_skirt_length = 0
notes = 
nozzle_diameter = 0.35,0.35
only_retract_when_crossing_perimeters = 1
output_filename_format = [input_filename_base].gcode
perimeter_acceleration = 6000
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 80
perimeters = 3
post_process = 
print_center = 160,140
raft_layers = 0
randomize_start = 
resolution = 0
retract_before_travel = 2,2
retract_layer_change = 1,1
retract_length = 1.0,1.0
retract_length_toolchange = 2,2
retract_lift = 0,0
retract_restart_extra = 0,0
retract_restart_extra_toolchange = 0,0
retract_speed = 20,20
rotate = 0
scale = 1
skirt_distance = 10
skirt_height = 1
skirts = 3
slowdown_below_layer_time = 15
small_perimeter_speed = 30
solid_fill_pattern = rectilinear
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extrusion_width = 0
solid_infill_speed = 120
spiral_vase = 0
start_gcode = G28 ; home all axes
support_material = 
support_material_angle = 0
support_material_enforce_layers = 10
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_layers = 1
support_material_interface_spacing = 0
support_material_pattern = rectilinear-grid
support_material_spacing = 3
support_material_speed = 40
support_material_threshold = 45
temperature = 180,200
threads = 4
toolchange_gcode = 
top_infill_extrusion_width = 100%
top_solid_infill_speed = 120
top_solid_layers = 5
travel_speed = 150
use_relative_e_distances = 0
vibration_limit = 0
wipe = 0,0
z_offset = 0
```

dexy31337, 2015-01-25:
1. Why your config has "# generated by Slic3r 0.9.10b" while it should be at least 1.1.7?
2. Please upload STL.
3. Dou you have "Don't support bridges" option on?
4. I don't like "fill_density = 0.4", it sems like 0.4%, so basically no fill at all, and in that case it it legit, that you have hollow model.

nothinman, 2015-01-30:
1. Because it's been generated with 0.9.10b?
3. I do, and this is it... thanks. Experimental option that's on by default, that's great...
4. No, 0.4 is 40%. Model was also fine, support was missing.

nothinman, 2015-01-30:
@alexrj Disable "don't support bridges" when support is turned on?

closed 2015-01-30 by nothinman

alexrj, 2015-01-30:
@nothinman, "Don't support bridges" only affects support material generation so disabling it when support is turned on doesn't make much sense.  :)