slic3r/Slic3r #2584: slic3r last update error slicing stl with untouch default config

c128, 2015-01-24:
./slic3r.pl 
Gtk-Message: (for origin information, set GTK_DEBUG): failed to retrieve property `GtkRange::activate-slider' of type `gboolean' from rc file value "((GString*) 0x5315f80)" of type `GString'
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)

slicing file
http://www.thingiverse.com/thing:771

after error need to reboot PC because very slow!

alexrj, 2015-01-25:
That file is very broken. Not even Netfabb is able to fix it. Slic3r can't do miracles...

alexrj, 2015-01-25:
(The error itself might be linked to #2569, but the output is garbage because the input is not valid)

closed 2015-01-25 by alexrj