slic3r/Slic3r #2585: Layers skipped in a manifold STL of a Prusa i3

DiThi, 2015-01-24:
http://i.imgur.com/HIxTA2b.png
http://i.imgur.com/cSI3HI2.png
http://i.imgur.com/prc0cKb.png

STL is "EJE X/Eje X izquierda.stl" in the Prusa i3 Hephestos kit https://static-bqreaders.s3.amazonaws.com/file/Prusa/piecess_Hephestos.zip

I prined it 3 times already thinking it was a hardware problem before I saw it skip actually skip the layers and then checked the generated gcode...

DiThi, 2015-01-24:
Oddly enough, I don't have that problem when calling Slic3r from Repetier-Host (that screenshot above is made loading the .gcode directly).

alexrj, 2015-01-25:
I confirm I can reproduce this with 1.2.6-dev and default settings.

alexrj, 2015-01-25:
I think this is related to the upstream bug in the Clipper library:
http://sourceforge.net/p/polyclipping/bugs/123/

alexrj, 2015-01-28:
Okay, I fixed this in b1f1893481c744801249d143971b6cf2d27308ef.
It was probably not very related to the mentioned Clipper bug.

[image]

Thank you for reporting!

closed 2015-01-28 by alexrj