slic3r/Slic3r #2586: STL crashes slic3r 1.2.5 on windows on send to printer

CptanPanic, 2015-01-25:
I have a STL that if I load into slic3r and try to send to octoprint, it crashes slic3r without actually sending it every time.  Below is a link to the stl in question.  

http://www.thingiverse.com/download:291008

alexrj, 2015-01-25:
Can you reproduce the crash with default settings?
Also, can you reproduce it by exporting G-code to a file instead of sending it to Octoprint?
Also, what operating system are you on?

CptanPanic, 2015-01-25:
Yes it happens with default settings, and it does happen when exporting as g-code as well.  I am running windows 8.1

CptanPanic, 2015-01-29:
Is there any workaround to this in the meantime?

lordofhyphens, 2015-01-29:
Other than don't use it? I think 1.2.6 will be out in the next day or so.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

CptanPanic, 2015-01-29:
I ask because this bug is only on the milestone list for 1.2.7 and not 1.2.6, so it may be another month or so.

alexrj, 2015-01-29:
I'll release 1.2.6 as soon as I recover from a flu that is keeping me in bed…
1.2.6 will contain the fix (that I assume is related to this report), and this is 1.2.7 milestone as a reminder to ask you whether it's fixed after 1.2.6 is released :)

alexrj, 2015-02-01:
Hello, can you confirm this is fixed in 1.2.6? Thank you!

CptanPanic, 2015-02-06:
Yes this fixes it.  Thanks.

closed 2015-02-06 by CptanPanic