slic3r/Slic3r #2588: 1.2.5 Unequal distance between perimeters

Taurus1-1, 2015-01-25:
Hello,

comparing two GCode files (one made with v1.2.5 and one made with v1.1.7) I found a different behavior concerning the perimeters.
The pictures below show each the 2nd layer. 
3 Perimenters: red, green, blue

In the GCode file made with 1.2.5 (upper picture) the distance between the perimeters is unequal. The distance between green and blue is bigger than between green and red.

In the GCode file made with 1.1.7 (lower picture) the distance between the perimeters is equal and the distance is smaller.

[image]

[image]

More details:
https://www.dropbox.com/sh/zpszs38js65wztg/AABosa7dOmkl4BC-nYgRhlpda
Linux Mint 17 x86

Peter

simonkuehling, 2015-01-26:
Hi Peter,

a recent change in the 1.2.x versions was a reduced extrusion width default for the external perimeter loop to aid in printing detailed surface texturing and fine features.

If you take a look into your GCODEs, right at the top the calculated extrusion widths are listed as a comment:

```
; generated by Slic3r 1.2.5 on 2015-01-25 at 15:55:09

; external perimeters extrusion width = 0.50mm      <=====================
; perimeters extrusion width = 0.85mm
; infill extrusion width = 1.02mm
; solid infill extrusion width = 0.85mm
; top infill extrusion width = 0.85mm
; support material extrusion width = 0.50mm
```

```
; generated by Slic3r 1.1.7 on 2015-01-25 at 16:40:54

; perimeters extrusion width = 0.50mm
; infill extrusion width = 1.02mm
; solid infill extrusion width = 0.85mm
; top infill extrusion width = 0.85mm
; support material extrusion width = 0.50mm
```

While this is intended behavior, it can be overwritten in the "Print Settings" tab -> "Advanced" -> "External Perimeters".
If you set a value of 100%, you will get the "old" 1.1.7 behavior where all perimeter loops have equal widths.

alexrj, 2015-01-26:
Thank you @simonkuehling for taking the time to answer :)

closed 2015-01-26 by alexrj