slic3r/Slic3r #2590: Multicore Processing - possible?

simonkuehling, 2015-01-27:
Hi @alexrj,

i am not sure if this question has been answered before - i didn't find an existing issue on that... 

i just noticed that slic3r seems to use only one core on my quad-core cpu currently. I am aware of limitations in some programming languages regarding multithreading on more than one core - not sure about perl though.
Is multicore processing something to consider for the future, or is something rendering it impossible with the current implementation?

Lord2Vader, 2015-01-27:
Advanced->Other->Threads
Slic3r uses 4 cores on mine

simonkuehling, 2015-01-27:
You're right - i refined my statement above. It is indeed using all of my 4 cores as well. But it seems to use only one core at a time, but alternating between them every few seconds... Most notably during support material generation...

Lord2Vader, 2015-01-27:
How can you see that? On my pc all 4 cores are 100% on slic3r

c128, 2015-01-27:
is possible add cuda(nvidia)  support ?

Some library used in slic3r can support cuda option.

pyrophreek, 2015-01-27:
How many threads do you have set? I find I have to set at least 16 before I start to max out my CPUs.  Try increasing the number of threads while you watch the CPU load, and when you hit 100% on all CPUS that is the optimum number.

simonkuehling, 2015-01-27:
@Lord2Vader, i am watching with gnome-system-monitor or htop on Ubuntu Linux 14.04 64bit

Ok, thanks to @pyrophreek we can call this resolved - i had only tried up to 9 threads on my quad-core system so far... increasing this to 16 pushed all cores to their limits almost throughout the complete slicing process.
i just did not expect a number this high to make it work at full throttle. 

Thanks all for your help!

closed 2015-01-27 by simonkuehling

pyrophreek, 2015-01-27:
Yeah I was surprised as well. I think it has to do with the way the processor handles hyperthreading/multithreading.  Also it depends on the number of bits/thread.

alexrj, 2015-01-27:
Not all steps are parallelizable - at some points the logic needs to return to a single thread. But this is going to improve in the near future because I'll be using C++ threads instead of Perl threads which are slower, and also because I plan to parallelize different steps (for example, support material and infill can be done at the same time because they're not dependent from each other).
I don't understand why it would need a value of 16 to reach 100% though… Is it actually faster than setting, say, the number of the cores you have? Or maybe it takes the same time but just reports 100% usage? :-)

whosawhatsis, 2015-01-27:
It might take more than the number of physical cores to max out the processor if those cores have hyper-threading.