slic3r/Slic3r #2593: 1.2.5 Ex: Support material too far away from high overhang curved surfaces

kyrreaa, 2015-01-28:
Printing a box with for example rounded edges towards the buildplate and using full support creates support material that is too far from the object to support it at all.

Will see if I can add some pictures to demonstrate this behavior.

alexrj, 2015-01-28:
Please read the guidelines and make sure your report complies with those. So far it's very incomplete

kyrreaa, 2015-01-28:
Images of a 30mm dia, 60mm long cylinder being sliced sideways show the support wrapping around the ends of the cylinder but the filament will just hang freely when printed. Support ends up around 3-5 layers below the surface it supports.
Maby due to distance being at an angle?
[image]
[image]
[image]

At 0.2mm layers and 0.4-0.5mm extrusion width on 0.4mm nozzle prints doesn't appear to touch the support at these rounded edges.

lordofhyphens, 2016-07-05:
Probably has something to do with the support width; could you set the support width to the thinnest possible (which is your nozzle diameter)?