slic3r/Slic3r #2596: 1.1.7 & 1.2.5 weird support generation

karrack, 2015-01-28:
the slicer versions i'm using seem to be never able to generate supports correctly. When slicing, slic3r switches to different layer heights which i can't even see a pattern. For instance whe i just take the 5_mm_calibration_steps and slice them with 40% honeycomb infill (infill percentage and pattern doesnt matter): http://www.thingiverse.com/thing:24238
(note: i unchecked "don't support bridges", but the issue will occur with every other support as well)

The first layer is correct (.3mm) but then it will generate a layer at .57mm where only support is printed. it goes on with a layer at 0.6mm (normal model layer height) witout support and then another layer at 0.9 consisting of support and the normal model. next layer is at 1.2mm (model only) followed by 1.27 (support only)
[image]
[image]
[image]

slic3r 1.1.7 has the same problems except i cant get such nice images

oschonrock, 2015-01-29:
I can confirm this behaviour of "skipping layers", but I assumed it was a "feature not a bug". ie the skipped layers were designed to make the support "more flimsy". 

It does result in very flimsy "stringy" support. The downside is that they are somewhat "compressable" and "heavy overhangs" will doop slightly.

Lenbok, 2015-02-01:
It is a feature not a bug. Inspired by: 
http://hydraraptor.blogspot.co.nz/2012/08/peel-able-support.html

alexrj, 2015-02-01:
Skipping layers… you're misinterpreting. No layers are skipped. You actually have TWO distinct piles of layers: the object ones, and the support ones. Incidentally, they have different heights. But each one of those two piles is *continuous* and each layer is firmly attached to a layer below.

Look at a 3D preview of the object you mentioned above:
[image]

As you see, there's no skipped layer.

Have you actually tried printing?

closed 2015-02-01 by alexrj