slic3r/Slic3r #2597: Missing infill on certain layers

zack112358, 2015-01-28:
I'm having a problem with a certain STL file. On certain layers, the infill is mostly absent. This occurs whether I try to use rectilinear, honeycomb, or 3D honeycomb infill.  The model features some very shallow slopes, some of which intersect into bridges. It looks like at least some of the affected layers are layers where bridging on those shallow slopes occurs.

My print settings and the STL file are here: https://gist.github.com/zack112358/4e6ab15169addaec122a. Problem occurs at Z=8.5, 9.3, 10.5, 12.9, 13.3, 16.1.

zack112358, 2015-01-28:
This is Slic3r 1.2.5.92, Mac OS X.

alexrj, 2015-01-28:
I confirm I can reproduce this on 1.2.6-dev.

alexrj, 2015-01-28:
Looks like it's the same issue as #2539

zack112358, 2015-01-28:
Oh, yeah, looks likely. Sorry about the dupe.

alexrj, 2015-01-28:
NP, we have one more test case.

alexrj, 2015-01-30:
Okay, fe4f79ba904d8fbfa277e6b6deb4f81286ac145c fixes this as well.

closed 2015-01-30 by alexrj

mrtn-t, 2019-05-26:
I have the same problem with Gyroid (3d honeycomb and other infils are working correct). 
Infil, I'm using latest , version 1.3.0 .
I tried with different percentage, but there is no difference. 
Small problems at the top from 1.8 to 3.6mm. Then HUGE gaps from 3.8 to 4.8mm. Then again from 8.2 to 11.2mm - almot half of infill is missing. 14.6 to 17.6 mm again.
STL file is in attached zip file.
Best regards, Martin
[planka2.zip](https://github.com/slic3r/Slic3r/files/3220081/planka2.zip)