slic3r/Slic3r #2598: First top solid layer is printed using bridge (speed and cooling) parameters....

daviofranke, 2015-01-29:
Hello,

This problem refers to version 1.1.7, I tested the 1.2.5 version and the problem remains the same.

Regardless of the speed of "top solid infill" and "cooling" settings, the first solid top layer is printed using the bridge settings of speed and cooling.

This fact causes problems of warping of parts and adhesion between layers, when the bridges need to be printed with "100% fan speed", since in that case when it arrives in the respective layer, the piece is suddenly cooled.

I can work around the problem, reducing the "bridge fan speed" to 50%, but in this case the bridges do not print as expected, since they require more cooling.

The pictures below show some unwanted results:

[image]

Layer affected by the problem, the printing was stopped because the piece taken off the bed.

[image]

In this, the warping caused an undesired result.

[image]

In this, it is evident adhesion failure between the layer with excessive cooling and the next.

Well, the speed does not cause problems, but the cooling should follow the general parameters and not bridges parameters. I think the ideal would be to use the bridge parameters only when the previous layer has not infill.

Also, I have a suggestion regarding the bridges:

- Cooling start or increase before the start of printing bridge, something like 1 second delay before, so that the ventilation has time to reach the desired speed.

- It is common for the bridge to be printed more slowly relative to the rest of the piece, then it would be interesting if there was a gradual slowdown before starting the printing of the bridge, as the abrupt transition speed not always promotes good adhesion between the bridge and the overlapping area.

I will be available for any clarification.

Thank you,

Davio Franke.

alexrj, 2015-01-29:
Yes, that happens by design and general consensus. That feature was discussed here before being implemented: #240

lordofhyphens, 2016-07-11:
This is by design. Closing.

closed 2016-07-11 by lordofhyphens