slic3r/Slic3r #2602: 1.2.5 moves acceleration adjustment - layer change

nicksears, 2015-01-30:
I'm having an "issue" with slic3r v 1.2.5. The issue is that the order of commands preceding the layer change has changed, but the underlying issue is that I cannot make the layer change gcode happen before the z move. For now I have to just use 1.1.7 (and it works fine). I'm not sure at which version this became "broken" for me.

I had been taking advantage of the fact that before a LAYER change (not just any z change like a hop) the G1 Z command will be preceded by [M204 S500 ; adjust acceleration] followed by the actual layer change [G1 Z0.200 F1500.000 ; move to next layer (0)].

To make this work I have a find and replace program (batch file utilizing the command line, it  recognizes "/n" as "next line") that finds:
"adjust acceleration/nG1 Z" 
This is the comment portion of the adjust acceleration line, and start of the layer z height change.
And it replaces it with:
"adjust acceleration/nINSERT/nWIPE/nCODES/HERE/nG1 Z"
This is because only layer changes have an acceleration adjustment preceding them.

The wipe protocol is a very long, concatenated set of commands, done in excel (I'm willing to share). It does a relative lift, moves to an absolute xy location, lifts to an absolute z, and then does a relative wipe movement (on toothbrushes). This is the issue - the wipe can start from any layer (height), must lift to the same height to wipe every time, but must return to the current layer height
. This is why it must happen before the layer change, the layer change gives it the appropriate z height that we've lost by mixing relative and absolute coordinates. The only other way around that is to attach the wiper to an axis on the side so it doesn't have to move to an absolute z height to wipe.

[slic3r 1 1 7 vs 1 2 5 layer change](https://cloud.githubusercontent.com/assets/6611900/5979300/36b882a2-a86c-11e4-92f1-616d8eaf957b.png)

I used excel to calculate a unit circle, scale it, and turn it into a "stadium" shape. This then brushes along two toothbrushes at the corner of my build plate; entering at one brush and exiting from another to maximize cleaning. I also have pre/post wipe commands that move onto and off of the brushes. I have excel file that then concatenates the commands (adding the "/n" after each) and I then insert it into the bat file.

There are a couple awesome features that could easily be added. Start by separating the layer change gcode into a before and after section. Also you could have a wipe enable option in each extruder in the printer settings. Give it a location (xyz), geometry or path, number of loops, and how often (every 1, 2,5,10 layers, etc). I don't have much actual coding experience (none in python) but I'd be glad to help.

alexrj, 2015-01-30:
Generating G-code with Excel sounds both funny and clever. :-P

Anyway, I just added an option for adding custom commands before layer change since that gets requested quite often. I also added a `[layer_z]` placeholder in layer change G-code that you can use for restoring Z after your wipe routine. So now you have two ways for doing what you need.

closed 2015-01-30 by alexrj

nicksears, 2015-02-02:
Awesome, [layer_z] sounds like it will let me do what i want, now I can put it in the layer change g-code!
Anyone have any thoughts on how to get it to do every x layers?
Anyone interested can find the find and replace program "fnr.exe" at "https://findandreplace.codeplex.com/" It wont let me upload excel files so email me for the wipe move calcs file or more details
[image]
.