slic3r/Slic3r #2603: Crash halfway through Gcode export (same file, one rotated) [1.2.5]

natevw, 2015-01-30:
Here's what I am doing:

1. Printing the file http://www.thingiverse.com/download:947608 ("Step_Blocks_-_Small.stl" of http://www.thingiverse.com/thing:589283)
2. Add it once to the plate, it is sideways
3. Add the *file* again to the plate (not just a second instance, otherwise step below rotates both)
4. Rotate the second file -90º around Y axis. [wanting to see which orientation prints best]
5. Auto-arrange parts if necessary
6. Export Gcode

I get this crash after Slic3r has written about half the layers:

```
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	slic3r.exe
  Application Version:	1.2.5.35
  Application Timestamp:	503fe867
  Fault Module Name:	XS.dll
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	54afd05d
  Exception Code:	c0000005
  Exception Offset:	000000000020f7e8
  OS Version:	6.1.7601.2.1.0.768.3
  Locale ID:	1033
  Additional Information 1:	53ca
  Additional Information 2:	53ca6a1eb975170a8d59149252e5999a
  Additional Information 3:	356d
  Additional Information 4:	356d51920ea0ce2a366b564297f6ad7f
```

natevw, 2015-01-30:
So actually, you can reproduce this without two copies of the file. It's the rotated version that crashes:

1. Load file in OP
2. Rotate -90º around Y-axis
3. Export Gcode [crashes as above]

alexrj, 2015-01-30:
Works on 1.2.6-dev.  I'm going to release 1.2.6 very soon.

imqqmi, 2015-02-01:
I experience a similar problem with the Standford Bunny:
http://www.inblik.nl/stanford_bunny.zip
The setting files are included in the zip.

- System 
  - Provider 
   [ Name]  Application Error 
  - EventID 1000 
   [ Qualifiers]  0 
   Level 2 
   Task 100 
   Keywords 0x80000000000000 
 
  - TimeCreated 
   [ SystemTime]  2015-02-01T12:32:43.000000000Z 
   EventRecordID 61170 
   Channel Application 
- EventData 
   slic3r.exe 
   1.2.5.76 
   503fe786 
   XS.dll 
   0.0.0.0 
   54afed71 
   c0000005 
   001bd87a 
   8c40 
   01d03e1b264e7452 
   D:\data\Projects\Prusai3V2\Software\Slic3r\slic3r.exe 
   D:\data\Projects\Prusai3V2\Software\Slic3r\lib\vrt\8027e0babc850e06667d7503a409ea98\XS.dll 
   6ab73f4a-aa0e-11e4-87d5-448a5b243e73

alexrj, 2015-02-01:
Hello, can you confirm this is fixed in 1.2.6? Thank you!

imqqmi, 2015-02-02:
Yes, it's fixed in 1.2.6! Thanks!

Alas it didn't fix the ears of the bunny, seems to be jagged after slicing. Cura does slice it properly.
http://www.inblik.nl/images/forumposts/Slicing.jpg

natevw, 2015-02-03:
Confirmed, can no longer reproduce with Slic3r 1.2.6 release. Thanks!

closed 2015-02-03 by natevw

alexrj, 2015-02-04:
Thank you for confirming!