slic3r/Slic3r #2605: 1.2.6-dev latest commit, crash if layerheight is 0 during editing of param

justmyopinion, 2015-01-31:
W7/x64
If Printing/layerheight is changed by clearing edit field an error shows in status (error in new_from_config_width) but if a 0 is added to empty field (ex 0.1) Slic3R crash and exits.

Any STL and  default settings.

alexrj, 2015-01-31:
I can't reproduce this. I loaded an STL file, then deleted the value in Print Settings -> Layer height
Are you sure you did Build.PL after git pull?

justmyopinion, 2015-01-31:
I did yes.
but I changed "First layer Height" obviously, sorry
Here is what happened:
[image]

I start "Perl Slic3R.pl" with default settings.
Drag STL file into plater
Enter Print settings/First Layerheighr. show 0.35
Backspace to clear numerical field.
Try to change FLH to 0.1 and crashes.

alexrj, 2015-02-01:
Ah okay, now I can reproduce it, thanks :-)

alexrj, 2015-02-01:
Fixed!  There was no validation for that field.

closed 2015-02-01 by alexrj

justmyopinion, 2015-02-01:
Verified to be solved with latest commit