slic3r/Slic3r #2606: [feature request] Manual modification of support

lukasz51, 2015-01-31:
You can add a manual modification of the support? As in Simplify3D.
https://www.youtube.com/watch?v=-v5GHJ2Qwyk  0:55min

alexrj, 2015-01-31:
"As in XYZ" – I don't like this, and it's not helpful. Let's first focus on *why*, and only after on *how*.

Can you more by elaborating on cases where you actually wanted to modify Slic3r's support? Of course I'm sure there are many. We need to study them before implementing anything.

lordofhyphens, 2015-02-01:
And is it better to do some of these supports in a general 3d modeling
program?

alexrj, 2015-02-01:
I'm completely open to interactive modification, which makes sense - but we need test cases to study

alexrj, 2015-02-01:
I don't like to do things just because they are good for marketing, regardless of how they'll be used ;)

RoaddogLabs, 2015-02-18:
I use manual support for cases where I don't need or want support in some areas but want it in others.  One reason I may not want the support is that for that particular part it may not need support in some areas that is auto generated but does in others.  Cleaning the support is time consuming and adds time to the print. 

For example I print  5-8 kg or so a week of printer parts.  For the Rework extruder I want support for the extruder idler and the hot end fan mount tabs.  I don't want it in the nut traps or mounting holes which is where the stable version of Slic3r puts it.  I use S3D for those parts. Another set of parts are the ends of a Smartrap. Even using auto mode Slic3r doesn't quite put the support where it needs support based on real world observations and not necessarily what the software thinks the part may need.  With some complex parts not all supports will require the same printing parameters as other support structures.  With the support algorithms (in all programs, not just Slic3er) sometimes there is too much support, sometimes not enough and sometimes not exactly in the right place. Manual supports give advanced users more flexibility.

@ lordofhyphens...  Designing parts with support structures or things like tabs should be done as a last resort. The purpose of the CAM is to aid in the manufacturing of the part.  By having to implement strategies in the actual part you limit that particular drawing to a single process, in this case additive manufacturing.    There are some instances where models are used across more than a single process and having to include what should be in the CAM in the original model adds unnecessarily to the workflow. Having to add the support structures to the original part drawing may also introduce issues using that part as an assembly component particularly in an application where one is modeling motion constraints or fit tolerance.

pixel3Design, 2015-04-11:
Totally agree. There's a python blender plugin to generate trees that can be useful to generate S3D like supports. Slic3r already has modifiers that can be used to generate supports only where needed. Indeed will be really useful having it in a more simplified and visually editable way. It will put Slic3r in a really new higher and professional level.

lordofhyphens, 2017-03-19:
Closing as linked issue is better described, making this a dupe.

closed 2017-03-19 by lordofhyphens