slic3r/Slic3r #2607: missing dependencies

swsch, 2015-02-01:
I've cleaned out /usr/local/lib/perl5 and let cpanm use ~/perl5.

Slic3r built, but the tests would only run successfully after I manually installed

  * strictures
  * Module::Runtime
  * Devel::GlobalDestruction
  * Import::Into
  * Role::Tiny

I'm not sure if they should be classified as "required" or "recommended", though.

alexrj, 2015-02-01:
Hm.  Actually those are not Slic3r direct dependencies but they are dependencies of Moo, and are normally installed by cpanm when Moo is installed...

swsch, 2015-02-01:
dang, another upstream thingie ... but wait for my next one :-)

closed 2015-02-01 by swsch

alexrj, 2015-02-01:
Heh :-)
But I think that only depends on something specific to your perl environment. Those dependencies *are* installed automatically most of the time...

gege2b, 2015-02-05:
Try::Tiny (a Moo dependency) was not automatically installed on my linux box

I've mentionned this here : https://github.com/alexrj/Slic3r/issues/2579#issuecomment-71374254
Anyway I've got no other dependency failure