slic3r/Slic3r #2609: Layers view stays empty

swsch, 2015-02-01:
The 3D preview is nice but a tad slow on my old box (now that it works :-), so I switched to the layers view, which was empty, not even the vertical slider was there.

After some digging around, I found that the following lines fix this problem for me:

```
diff --git a/lib/Slic3r/GUI/Plater.pm b/lib/Slic3r/GUI/Plater.pm
index de23e40..f743b06 100644
--- a/lib/Slic3r/GUI/Plater.pm
+++ b/lib/Slic3r/GUI/Plater.pm
@@ -117,9 +117,13 @@ sub new {
     if ($Slic3r::GUI::have_OpenGL) {
         $self->{toolpaths2D} = Slic3r::GUI::Plater::2DToolpaths->new($self->{preview_notebook}, $self->{print});
         $self->{preview_notebook}->AddPage($self->{toolpaths2D}, 'Layers');
+        $self->{toolpaths2D_page_idx} = $self->{preview_notebook}->GetPageCount-1;
     }
-    
+
     EVT_NOTEBOOK_PAGE_CHANGED($self, $self->{preview_notebook}, sub {
+        if ($self->{preview_notebook}->GetSelection == $self->{toolpaths2D_page_idx}) {
+            $self->{toolpaths2D}->reload_print;
+        }
         if ($self->{preview_notebook}->GetSelection == $self->{preview3D_page_idx}) {
             $self->{preview3D}->load_print;
         }
```

swsch, 2015-02-01:
After rebuilding perl (with threads) and Slic3r, I see that this problem does not come up with background processing enabled.

alexrj, 2015-02-02:
You didn't have threads or background processing before rebuilding?

swsch, 2015-02-02:
Well, what can you expect from a box where perl has been compiled without threading support?

Never missed it before Slic3r :-)

alexrj, 2015-02-02:
Aha, okay! That was it then :-)

closed 2015-02-02 by alexrj