slic3r/Slic3r #2610: Infill artifacts under vertical extrusions

Vicious-one, 2015-02-01:
Somehow, even with solid infill area threshold low enough, solid infill layers have gaps under vertical extrusions (near the edges):
[image]

Extruded parts:
[image]

STL and config:
http://vserv.sinp.msu.ru/temp/infill-gaps.tar.gz

Vicious-one, 2015-02-02:
The gaps are not generated on this model when "extra perimeters" are disabled.

Vicious-one, 2015-02-16:
The problem became much more aggravated with a recent commit:
[image]

alexrj, 2015-02-23:
Fixed as well.

commit 45c91b2a (2015-02-23): Don't trigger extra perimeters when less than 30% of the upper loops would benefit from it. #2664. Also fixes #2610
closed 2015-02-23 by alexrj