slic3r/Slic3r #2611: Incorrect geometry based on nozzle size Slic3r 1.1.7, Slic3r 1.2.6

bstory2001, 2015-02-01:
Wwhen trying to print this small gear with a 0.5mm nozzle, the size of the teeth are reduced.  In fact they get progressively smaller when the stl is sliced with nozzle sizes from 0.35 mm up to 0.5 mm as seen in the animation below.  Measuring the STL, the teeth on the gear protrude roughly 0.9 mm. The gear is built into this part: 

( https://github.com/firepick1/FPD-LooseCanon/blob/dev/mechanical/3DLCxxxx%20-%20LooseCanon%20parts/STL/3DLC0002S%20Delta%20pulley%20(150T).stl ).  

Note that I get the same result if I slice the whole pulley or split the gear out separately using Slic3r split feature. 

The nozzle sizes in the gif below are 0.35(largest teeth), 0.38, 0.42, 0.46, 0.50(smallest teeth).  
[image]

Slicing the file using Cura produces the correct geometry as seen in the bottom image (Cura top left, Slicer top right.)  The bottom sections are images of the SLT.
[image]

*Screenshots from Repetier Host GCode Preview.  config.ini below

```
# generated by Slic3r 1.2.6 on Sun Feb  1 12:29:54 2015
avoid_crossing_perimeters = 0
bed_shape = 0x0,210x0,210x210,0x210
bed_temperature = 0
before_layer_gcode = 
bottom_solid_layers = 2
bridge_acceleration = 1000
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 100
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 600
disable_fan_first_layers = 5
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = M104 S0 ; turn off temperature\nM140 S0; turn off bed temp\nG28 X0  ; home X axis\nM84     ; disable motors
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 70%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0
fan_always_on = 1
fan_below_layer_time = 20
filament_diameter = 3
fill_angle = 45
fill_density = 20%
fill_pattern = honeycomb
first_layer_acceleration = 100
first_layer_bed_temperature = 0
first_layer_extrusion_width = 200%
first_layer_height = 0.35
first_layer_speed = 50%
first_layer_temperature = 0
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 1000
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 15%
infill_speed = 80
interface_shells = 0
layer_gcode = 
layer_height = 0.2
max_fan_speed = 100
min_fan_speed = 35
min_print_speed = 10
min_skirt_length = 0
notes = 
nozzle_diameter = 0.5
octoprint_apikey = 
octoprint_host = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 150
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 80
perimeters = 1
post_process = 
pressure_advance = 0
raft_layers = 0
resolution = 0
retract_before_travel = 2
retract_layer_change = 1
retract_length = 1
retract_length_toolchange = 10
retract_lift = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 30
seam_position = aligned
skirt_distance = 0
skirt_height = 1
skirts = 0
slowdown_below_layer_time = 10
small_perimeter_speed = 80%
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 80%
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = G28 ; home all axes\n
support_material = 0
support_material_angle = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 100
support_material_threshold = 0
temperature = 0
thin_walls = 1
threads = 2
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 70%
top_solid_layers = 2
travel_speed = 130
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 10
wipe = 0
xy_size_compensation = 0
z_offset = 0
```

Lord2Vader, 2015-02-01:
Well repetierhost shows the line where the nozzle is at. But at an 0,5mm nozzle the nozzle has to go in for 0.25mm ;) Seems right to me!

If your printing a radius 50mm your nozzle does a radius of 49,75mm.

Try Outer perimeters 0.1mm and it should not be that great in difference, just like choosing a smaller nozzle

bstory2001, 2015-02-01:
The actual print has the lack of teeth, not just the 3d display.  Either way, I changed the external perimeter to 0.35mm and the teeth come out much better.  Thanks :)

alexrj, 2015-02-02:
Yes, that's correct behavior. One *feature* of Slic3r is the collision check. When your nozzle (or, better, your extrusion width) is too large for extruding very tiny spikes, they are skipped to prevent over extrusion. You can still set a thinner extrusion width manually. Or you can disable collision check by unchecking the "Detect thin walls" option (yes, this is not documented).
There's a recent blog post by Richrap about adjusting extrusion width.

Also, Repetier's 3D rendering is very fake since it doesn't know the true extrusion widths. If you look at the new 3D preview in Slic3r itself you'll see a much saner slideshow…

alexrj, 2015-02-02:
While this is correct behavior, and collision checking is a feature (that can be disabled), I'm still unhappy since most users would think that this is a bug.   For one person that reports a potential issue here, there are hundreds outside that just think "this doesn't work".

So I really would like to think how to prevent this misunderstanding in terms of user experience. Maybe we should partially disable collision checking.

Vicious-one, 2015-02-02:
@alexrj -- Maybe a more apt name or tooltip would suffice for that particular option

Lord2Vader, 2015-02-02:
Or maybe just a tip in dokumentation called "How plastic is layed down" with things like "How true size is accomplished"?

bstory2001, 2015-02-03:
Color coding the preview is probably a pain.  Would it be easy to implement some feedback to let the user know when some automatic setting is effecting the output and give them an idea of what to look at in the preview?  

Maybe some info down in the Info section to indicate things such as
Object has:  bridging, thin walls detected, extra perimeters needed, overhangs, or anything else that may encourage the user to reconsider his/her settings.