slic3r/Slic3r #2613: Slic3r >=1.2.5 (at least) crashes in GUI/Tab.pm on linux x86_64

swsch, 2015-02-01:
* run bin/slic3r (from 1.2.5 release) or perl slic3r.pl (from git)
  * click on tab "Print Settings"
  * click into field "Solid layers/Top" and enter a new value via keyboard
  * click on "Infill" in tree view

Slic3r 1.2.5 crashes with 

`Can't call method "get_field" on an undefined value at .../GUI/Tab.pm line 405`

Slic3r from git crashes with

`Can't call method "get_field" on an undefined value at .../GUI/Tab.pm line 414`

If you use the spin buttons or leave the field with tab before clicking into the tree view, Slic3r continues to work.

alexrj, 2015-02-02:
Interesting, thank you - this doesn't happen on OS X.  Maybe this is related to the other Linux bug.

simonkuehling, 2015-03-11:
Just realized i posted a duplicate of this at #2729.

As i mentioned there, i could not reproduce my error. Same applies for the steps mentioned by @swsch - those work without a fault here on ubuntu 14.04 with the latest 1.2.7-dev from git... really weird behavior :-)

alexrj, 2015-03-27:
I can't reproduce this on Linux as well... :(

alexrj, 2015-03-27:
@swsch, can you reproduce this with current git?
If so, what's the output of `perl slic3r.pl --debug`? (I'm interested in the Wx version)

swsch, 2015-03-27:
I did a `git pull`, followed by `perl Build.PL` and `perl Build.PL --gui`, which is where we join the process:

```
$ perl Build.PL --gui
App::cpanminus is up to date. (1.7027)
Class::Accessor is up to date. (0.34)
Growl::GNTP is up to date. (0.20)
LWP::UserAgent is up to date. (6.13)
Net::Bonjour is up to date. (0.96)
OpenGL is up to date. (0.6704)
Socket is up to date. (2.018)
--> Working on Wx
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Wx-0.9927.tar.gz ... OK
Configuring Wx-0.9927 ... OK
Building and testing Wx-0.9927 ... OK
Successfully installed Wx-0.9927 (upgraded from 0.9923)
1 distribution installed
Wx::GLCanvas is up to date. (0.09)
$ ./slic3r.pl --debug                                                  255   master 
wxWidgets version wxWidgets 3.0.0, Wx version 0.9927
Data directory: /home/stefan/.Slic3r
Can't call method "get_field" on an undefined value at /home/stefan/repos/Slic3r/lib/Slic3r/GUI/Tab.pm line 414.
```

After starting slic3r, I repeated the steps above and got the same error.

So I added a bit of debugf:

```
diff --git a/lib/Slic3r/GUI/Tab.pm b/lib/Slic3r/GUI/Tab.pm
index 00eb1d6..ba37bec 100644
--- a/lib/Slic3r/GUI/Tab.pm
+++ b/lib/Slic3r/GUI/Tab.pm
@@ -410,6 +410,11 @@ sub get_preset_config {
 sub get_field {
     my ($self, $opt_key, $opt_index) = @_;
     
+    Slic3r::debugf "%s: %d\n", $opt_key || "no_key", $opt_index || -1;
+    foreach my $page (@{ $self->{pages} }) {
+        Slic3r::debugf "  %s\n", $page->{title} || "untitled_page";
+    }
+    Slic3r::debugf "\n";
     foreach my $page (@{ $self->{pages} }) {
         my $field = $page->get_field($opt_key, $opt_index);
         return $field if defined $field;
```
... and performed my test again. The last lines are:

```
extruder_clearance_radius: -1
  Layers and perimeters
  Infill
  Skirt and brim
  Support material
  Speed
  Multiple Extruders
  Advanced
  Output options
  Notes

extruder_clearance_height: -1
  Layers and perimeters
  Infill
  Skirt and brim
  Support material
  Speed
  Multiple Extruders
  Advanced
  Output options
  Notes

standby_temperature_delta: -1
  Layers and perimeters
  Infill
  Skirt and brim
  Support material
  Speed
  Multiple Extruders
  Advanced
  Output options
  Notes

extra_perimeters: -1
  untitled_page
  Infill
  Skirt and brim
  Support material
  Speed
  Multiple Extruders
  Advanced
  Output options
  Notes
```

The first three items appear at the end of the `_update` sub around line 820, the final `get_field` for `extra_perimeters`, which causes the exception, happens at the start of another invocation of the same sub `_update` as a bit more peeking and poking shows.

I have no idea why two calls to `_update` happen with different contents of `$self->{pages}` in this situation.

alexrj, 2015-05-04:
@swsch, thank you for the debugging. I guess this might be related to some wxWidgets 3.0.0 behavior. I use an older version on Linux. Many Linux distributions still ship 2.8.

lordofhyphens, 2015-05-04:
Debian 8 (jessie) ships 3.0 and does not have 2.8 anymore. It's also
Stable, so this probably needs to be worked out.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

rokmarko, 2015-05-28:
Slic3r starts in simple mode but crashes in expert.
Linux Kubuntu 14.04, wx 3.0

alexrj, 2015-06-01:
Good news!  I was able to reproduce this on OS X as well and I fixed it. Nasty concurrency issue.

closed 2015-06-01 by alexrj

swsch, 2015-06-05:
works fine on linux x64, too.

Thanks a lot!