slic3r/Slic3r #2615: 1.2.6 pre-compiled for Windows x64 crashing with any STL

thiagopeixoto16, 2015-02-02:
All STL files are crashing 1.2.6. The software seems to close at the time to slice support (even with support disabled).
I'm using the very standard settings, the ones who comes with in the ZIP.
Anyway here find below the STL and INI file:
https://www.dropbox.com/sh/rj9g5q3kmwpptv8/AACvQzZRuF-fRdkv80DIjt63a?dl=0

This is a Windows 7 x64 machine.

Please let me know if you can reproduce the error.

Thanks!

Aerospacesmith, 2015-02-02:
I was able to slice without any error with your config.ini and stl in 1.2.6

thiagopeixoto16, 2015-02-02:
Updated ini file with the real problematic config.

I did some more tests and this is a kinda weird. 
The problem only happens if I open Slic3r with the .ini attached loaded,
In other hand If I open using the standard configs and then change to this attached setup the problem don't happen...

I tried with both x64 and x84 version running on Win7x64 with same behavior.

Please let me know how to help.

(BTW: The 3D slice preview is fanTAStic! )

Lord2Vader, 2015-02-05:
Same with me. Got some small objects loaded which were from the old 123D version. Bigger ones dont load and all of the new 123D versions dont work even small ones.

Tryed repair stl didnt help.
I cant slice them all with repetierhost and newest slic3r version 1.2.6

Lord2Vader, 2015-02-05:
Cant edit via mobile on github sry:
I use win7 x64 and both 32 and 64bit versions dont work.
And i meant. I CAN slice with slic3r and repetierhost. Just cant load it in slic3r

thiagopeixoto16, 2015-02-06:
Right, no problems via command.

thiagopeixoto16, 2015-02-09:
Since I have Repetier opened even working via Slic3r Plater I was able to get some error info, here it go:

15:24:25.313 : <Slic3r> Attempt to free unreferenced scalar: SV 0x17e4d558, Perl interpreter: 0x1140c0b8 at D:/3DPrinter/slic3r-mswin-x64-1-2-6-experimental/lib/std/threads.pm line 101.

thiagopeixoto16, 2015-02-09:
Another one. This is exactly at the time it crashes:

15:32:01.277 : <Slic3r> Attempt to free unreferenced scalar: SV 0x11042e48, Perl interpreter: 0x17e4d9e8 at D:/3DPrinter/slic3r-mswin-x64-1-2-6-experimental/lib/std/Slic3r.pm line 116.

tomgeor, 2015-02-24:
I can reproduce this error every time on 64-bit Fedora Linux 21, regardless of the STL file I choose. 

$ ./slic3r 
Attempt to free unreferenced scalar: SV 0x51c8790, Perl interpreter: 0x4632580 at /opt/3dp/Slic3r/lib/std/Slic3r/GUI.pm line 262.
Attempt to free unreferenced scalar: SV 0x51c8760, Perl interpreter: 0x4632580 at /opt/3dp/Slic3r/lib/std/Slic3r/GUI.pm line 262.
Attempt to free unreferenced scalar: SV 0x5a48ec0, Perl interpreter: 0x62e1270 at /opt/3dp/Slic3r/lib/std/Slic3r.pm line 116.

alexrj, 2015-06-01:
@thiagopeixoto16 and @Lord2Vader: this should be fixed as of 1.2.7.  Can you please test on that? Can we close this one? Thank you!

thiagopeixoto16, 2015-06-11:
I tested on the amazing 1.2.8 x64 version and that's fine. For me this can be closed. Thanks!

closed 2015-06-12 by alexrj