slic3r/Slic3r #2616: 1.2.6 post-processing scripts error

Aerospacesmith, 2015-02-02:
Running post-processing scripts in 1.2.6 gives an error "the configured post-processing script is not executable: check permissions". 
In 1.2.4, the script runs with no issues.
Trying to run the filament-weight script with perl 5.20

alexrj, 2015-02-02:
Uhm, what operating system?

Lenbok, 2015-02-03:
I got the same thing on linux, but when I put in the full path to the postprocessing script and saved the settings it is working again.

Aerospacesmith, 2015-02-03:
i have the full path specified. Here is my output settings: [image]

DanNsk, 2015-03-02:
I have the issue when i keep script path in quotes, if i remove quotes - that fixes the issue (same config worked in 1.2.4 i think)

alexrj, 2015-03-02:
Okay, I fixed this, thank you.

(@DanNsk, that's another thing - you need to remove quotes)

closed 2015-03-02 by alexrj