slic3r/Slic3r #2617: Infill pattern conforming to the object's shape?

whosawhatsis, 2015-02-02:
I've just been thinking about how to optimize infill, conforming to the shape of the object. For thin features, where infill is less than double the normal extrusion width, the space should be filled entirely, and the best way to fill is to move the nozzle down the medial axis, adjusting the extrusion width with each segment in order to fill the space. For filling larger spaces, why not use a square wave pattern like rectilinear fill, but always crossing perpendicular to the medial axis. This would produce a pattern that would conform to the shape of the object. It would be better for supporting the outlines of sloped surfaces, and also would be ideal for being covered with a concentric fill pattern. Alternating this pattern with a concentric sparse fill would create a grid that conforms to the print's shape. This pattern would also be ideal in many cases for use as a support structure to break away. Would this type of pattern be difficult to implement, given that Slic3r already has code for identifying the medial axis?

alexrj, 2015-02-02:
Interesting idea… Very interesting… That would avoid the problem with clipped rectilinear being ugly at some angles.

Regarding the first case, Slic3r already fills gaps with medial axis and adjusted extrusion width (although some cases are still not handled, but that's fixable).

whosawhatsis, 2015-02-02:
Yeah, I know that Slic3r already has code to do the medial axis gap fill for thin areas. I got the idea when thinking about how to extend this functionality to fill thicker areas.

lordofhyphens, 2016-06-28:
Is this covered by the variable-width gap fill logic in 1.3.0-dev?