slic3r/Slic3r #2618: 1.2.6 Wrong support generated

Zuborg, 2015-02-02:
This problem happened if I tilt this complex part (for produce lower quantity of a support plastic).

The first picture present a corner item in a "free space" without any link to the part and without support. This failure is same for any type of support (pillars, honeycomb, etc)
+45 degrees about X axis:
[image]

Here problem of a "pillars support": At left side of picture there are top items of pillars without bottom columns. We can see a raft for these pillars at the first layer but no more. At right there is very thin pillar which will be broken with its weight during printing.
-45 degrees about X axis:
[image]

source stl: https://drive.google.com/file/d/0BwiHfZggx8jHQzNvWENScW1YZ0E/view?usp=sharing

config.ini: https://drive.google.com/file/d/0BwiHfZggx8jHTTl2NUozYW9VUnM/view?usp=sharing

alexrj, 2015-02-02:
Yeah, definitely ugly support there.  I think pillars is not the best choice at the moment for that kind of model...

Zuborg, 2015-02-02:
I think really should be used cautiously "pillars" for complex models, especially where the result of the generation of support is difficult to see. However, the problem at the first picture is more serious, as it is not dependent of the type of support.

markfinn, 2015-02-06:
The first picture issue may be what I'm reporting in #2635

MikeFitz73, 2015-02-19:
I also have this same problem in the latest 1.26 release.  if I turn my object 45 degrees on x axis for example, support is missing in many areas.
I would like to know if we could add a feature.... would it be possible to be able to setup support Manually, let the user put support in where the user sees necessary?  Like in simplify3d? thank you.
Mike

lordofhyphens, 2017-03-19:
Looks fixed.
Rectilinear
[image]

pillars:
[image]

The narrow pillars are still an issue, but that's an issue of coverage with pillars (those thin areas are the attempt to cover areas that the pillars didn't get).

closed 2017-03-19 by lordofhyphens