slic3r/Slic3r #2619: [Feature Request] 1.2.6-dev: Add option to connect support sections with a pillar.

lordofhyphens, 2015-02-02:
One of the pieces I am trying to print with support has a very long, narrow overhang. 

Slic3r's support generation is calculating a few very tiny pillars that can easily get knocked loose from the bed. Is it feasible to add an option to try to connect tiny "islands" (say, less than 10mm x 10mm in area) to each other to help them stick better?

alexrj, 2015-02-02:
Screenshot?    (I heard of this request several times, just want to see the particular situation)
Isn't the base flange enough to connect them?

lordofhyphens, 2015-02-02:
[image]

There's the 3d path preview. Obviously if your bed adherence is awesome, it's not a problem. Some of us don't have awesome bed adherence all the time though. 

It probably doesn't need more than 2-3 layers of connection, but this is best left as a configurable item.

lordofhyphens, 2017-02-08:
@bubnikv your support code addresses this, doesn't it?

bubnikv, 2017-02-08:
Not exactly. My support code works just differently. I have some tricks in my sleeve to reduce the support density, but that is not implemented yet.

I have currently disabled the pillars in the prusa3d fork completely. The current pillars are too unreliable.