slic3r/Slic3r #2620: [1.2.6] Infill/perimeter overlap is not applied with modifier meshes

Vicious-one, 2015-02-02:
I've noticed that infill pattern, etc. can be changed with a modifier mesh, but overriding the infill overlap has no effect whatsoever.

alexrj, 2015-02-04:
Well, modifier meshes work by isolating a portion of the object and generating its own perimeters+infill. In this case, overlap is applied to the combo.

However if the modifier mesh and the main object have the same perimeter settings, a single set of perimeters is generated around both and modifier mesh only defines a portion of the infill. In this case, only the overlap of the main object is used.  I'll look into the possibility of using both overlaps in this case, but I'm afraid that's a good amount of work.

Does this answer your report?

Vicious-one, 2015-02-04:
Yes, it does (kind of). 
Anyway, this hint allows to accomplish what was intended (enhancing perimeter stitching with sparse infill). 
Closing.

closed 2015-02-04 by Vicious-one

Vicious-one, 2015-02-04:
Wow, a modifier with [a little] different perimeter parameters also allows to create solid thick walls inside a sparse infill and segment circular perimeters to enhance stitching. And that is a goldmine of neat strength-enhancing tricks. Thanks again!

alexrj, 2015-02-04:
If you happen to have a test case with pictures, we could make a little tutorial-blog post on that :)

Vicious-one, 2015-02-11:
Well, I do.

Here's a spoolholder with a bearing that I wanted to be light and a little flexy for bearing insertion, but quite rigid nonetheless.
[image]

Infill settings:
[image]

So I've created a cross-shaped modifier mesh whish is a little narrower than the model's walls and changed its infill and perimeter settings:
[image]

This results in proper sparse infill where needed: 
[image]

And sparse infill with rigid rips where appropriate:
[image]